carla-simulator/carla #1606: [0.9.4] inconsistent weather visual effects

dianchen96, 2019-05-03:
Hi!

I was working with CARLA 0.9.4 and found that the weather effects for the rainy weathers (HardRainNoon etc.) seem different depending on where we set the weather in the Python code. 
If I set the weather right after getting the the world, the raining effects seems missing. However the raining effects are normal if I set the weather after spawning vehicles, walkers etc.. Has anyone encountered similar issues or know why this happens? 

Set weather before spawning
[image]

Set weather after spawning
[image]
 The code snippet I was working on is
```
    settings = world.get_settings()
    settings.synchronous_mode = True
    world.apply_settings(settings)
    
    # set weather 1
    # world.set_weather(PRESET_WEATHERS[args.weather])
    
    blueprint_library = world.get_blueprint_library()
    
    vehicle_bp = random.choice(blueprint_library.filter('vehicle.ford.mustang'))
    vehicle_bp.set_attribute('role_name', 'hero')
    
    carla_map = world.get_map()
    
    actor_list = []
    
    # Intialize player
    start_pose = carla_map.get_spawn_points()[args.start_pose]
    end_pose = carla_map.get_spawn_points()[args.target_pose]

    player = world.spawn_actor(vehicle_bp, start_pose)
    player.set_autopilot(False)
    actor_list.append(player)
    
    # Spawn cars
    for _ in range(args.num_vehicles):
        bp = random.choice(blueprint_library.filter('vehicle.*'))
        while True:
            vehicle = world.try_spawn_actor(bp, random.choice(carla_map.get_spawn_points()))
            if vehicle is not None:
                vehicle.set_autopilot(True)
                actor_list.append(vehicle)
                print ("Spawned vehicle %d"%_)
                break
    
    # Add camera
    rgb_camera_bp = blueprint_library.find('sensor.camera.rgb')
    rgb_camera_bp.set_attribute('image_size_x', '384')
    rgb_camera_bp.set_attribute('image_size_y', '160')
    rgb_camera_bp.set_attribute('fov', '90')
    rgb_camera = world.spawn_actor(
        rgb_camera_bp,
        carla.Transform(carla.Location(x=2.0, z=1.4), carla.Rotation(pitch=0)),
        attach_to=player)
    actor_list.append(rgb_camera)
    
    # Set camera queues
    rgb_queue = queue.Queue()
    rgb_camera.listen(rgb_queue.put)
    
    # Add collision sensor
    crashed = -1
    def on_collision(event):
        nonlocal crashed
        # print ("crashed!!!")
        crashed = event.frame_number
        
    collision_sensor_bp = blueprint_library.find('sensor.other.collision')
    collision_sensor = world.spawn_actor(
        collision_sensor_bp,
        carla.Transform(),
        attach_to=player
    )
   
    #set weather #2
    # world.set_weather(PRESET_WEATHERS[weather_id])
```

germanros1987, 2020-06-26:
There is an issue related to this. I believe @XGodina told me about it. Xavi, could you take care of this?

XGodina, 2020-06-29:
Hi, @dianchen96. Currently, we have a weather system that the particles only spawn in front of the vehicle's camera.  According to those parameters like rain only spawn it when you run the script manual_control.py or spawn_npc.py. The reason for that is that if we generate particles on the whole map the performance is negatively affected.

Madecu, 2020-07-13:
Hi, @dianchen96. We have fixed it. Now the rain starts at the beginning and it won't disappear when you spawn another vehicle or change the camera.

dianchen96, 2020-07-15:
Awesome, thanks a lot!

closed 2021-03-22 by Madecu