carla-simulator/carla #1609: Highway Lane change

lijecaru, 2019-05-03:
How to perform lane changes in Carla? I am able to follow predetermined waypoints, but is there a way to make my vehicle override the next desired waypoint to change lanes? Also, if I spawn several vehicles on the highway, and I manually control one to go slower than the other vehicles, the following cars just stop instead of passing my car. Is there a way to change that behavior?

Derekabc, 2019-05-11:
Hi, @lijecaru, do you know how to lead my car running on the particular path, for example, turn left and merge path to the left lane. I found some information on the _scenario_runner_, but it is still not enough for me to figure out how to do that. Thanks

closed 2019-08-02 by stale[bot]

bapukokare, 2022-08-01:
> Hi, @lijecaru, how to follow predetermined waypoints and get a list of waypoints? Thanks

used carla_debug mode  and printed road id and append waypoint list on that road id 
import glob
import os
import random
import sys
import numpy as np
import time 

try:
    sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass

import carla

actor_list =[]

try:

    # Carla host
    client = carla.Client('localhost',  2000)
    client.set_timeout(20)
    #world = client.load_world()
    #client.reload_world()
    
    world = client.get_world()

    blueprint_library = world.get_blueprint_library()
    #get specific vehicle 
    vehicle1 = blueprint_library.filter('model3')[0]
    print(vehicle1)

    waypoints = client.get_world().get_map().generate_waypoints(10.0)
    filtered_waypoints = []
    for way in waypoints:
        #str(way.road_id), str(way.lane_id)
        # print road id by passing ,way.road_id)
        client.get_world().debug.draw_string(way.transform.location, str(way.road_id), draw_shadow=True, color=carla.Color(r=0, g=0, b=0), life_time=360.0,persistent_lines=True)
        if(way.road_id == 12 ):
            print(way.transform.location)
            filtered_waypoints.append(way)

    
    spawn_point = filtered_waypoints[0].transform # Car 1 start point
    spawn_point.location.z += 0.7
    actor1 = client.get_world().spawn_actor(vehicle1, spawn_point)
    print(actor1.get_transform().rotation.yaw)
    

    
    spectator = world.get_spectator()

   
    control = carla.VehicleControl()
    control.throttle = 0.3
    actor1.set_autopilot(True)

    #actor_list[0].apply_control(control)

    actor_list.append(actor1)
    time.sleep(20) # run program for 20 second 

finally:
    for actor1 in actor_list:
        actor1.destroy()
    print('\n done ')