carla-simulator/carla #1611: Use generate_map.py to import .fbx files exported by Unity?

0797liyn, 2019-05-06:
I want to know the generate_map.py is only work for the exported files of RoadRunner?
when i put my .fbx files(exported by Unity) in the folder RoadRunnerFiles/nedlmap01, and run generate_map.py, it is not work well. Is the problem of my  .fbx files?or other things?
Thank you for help.

> aglserver@aglserver01:~/Carla/test_basecarla/PythonAPI/util$ ./generate_map.py --force --usecarlamats
Found map in fbx folder: nedlmap01
Increasing per-process limit of core file size to infinity.
LogConsoleResponse: Display: Failed to find resolution value strings in scalability ini. Falling back to default.
LogInit: Display: Running engine for game: CarlaUE4
LogPakFile: Registered encryption key '00000000000000000000000000000000': 0 pak files mounted, 0 remain pending
LogPlatformFile: Not using cached read wrapper
LogTaskGraph: Started task graph with 5 named threads and 26 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.119906
LogICUInternationalization: ICU TimeZone Detection - Raw Offset: +8:00, Platform Override: ''
...
[2019.05.06-01.47.15:657][  0]LogInit: Display: 
Execution of commandlet took:  0.00 seconds
[2019.05.06-01.47.15:729][  0]LogShaderCompilers: Display: Shaders left to compile 0
Exiting abnormally (error code: 1)
done.
Traceback (most recent call last):
  File "./generate_map.py", line 189, in <module>
    main()
  File "./generate_map.py", line 26, in main
    generate_all_maps_but_list([])
  File "./generate_map.py", line 50, in generate_all_maps_but_list
    import_assets_commandlet(map_name)
  File "./generate_map.py", line 100, in import_assets_commandlet
    invoke_commandlet(commandlet_name, commandlet_arguments)
  File "./generate_map.py", line 145, in invoke_commandlet
    subprocess.check_call([full_command], shell=True)
  File "/usr/lib/python3.6/subprocess.py", line 291, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/aglserver/Carla/UnrealEngine_4.21/Engine/Binaries/Linux/UE4Editor /home/aglserver/Carla/test_basecarla/PythonAPI/util/../../Unreal/CarlaUE4/CarlaUE4.uproject -run=ImportAssets -importSettings="/home/aglserver/Carla/test_basecarla/PythonAPI/util/importsetting.json" -AllowCommandletRendering -nosourcecontrol -replaceexisting']' returned non-zero exit status 1.

p-schulz, 2019-05-14:
In general, I think you can import .fbx files straight to your Unreal Editor/Carla Assets folder, as this file format is best suited for Unreal. 

My guess by looking into the generate_map.py script (just open it with a editor), is that the script further expects an .xodr file - hence the OpenDRIVE file that comes with a RoadRunner generated map. It contains the logical data for your agents, such as lanes and stuff. So the script could basically work only with RoadRunner maps, or .fbx files where you also have an .xodr file.

Try importing your .fbx straight to Unreal and then place some RoutePlanner nodes on your streets.

0797liyn, 2019-05-24:
Thank you for you replay. I will try.

closed 2019-05-24 by 0797liyn