carla-simulator/carla #1613: automatic_control Basic option Car is not moving

guochun7648, 2019-05-06:
Hello,
I am new to Carla. I am trying to modify automatic_control.py in order to make it start at one location and end to another location. I have changed the destination location to (5.6, -88.9, 0) in the map Town03. However, after I spawn the car, it only works for W S on keyboard and it didn't move automatically at all. What's wrong with it? Please help me. Thank you.

 if args.agent == "Roaming":
            agent = RoamingAgent(world.player)
        else:
            agent = BasicAgent(world.player)
            spawn_point = carla.Transform(carla.Location(x=5.6, y=-88.9, z=2.5))
            agent.set_destination((spawn_point.location.x,
                                   spawn_point.location.y,
                                   spawn_point.location.z))

Derekabc, 2019-05-07:
https://github.com/carla-simulator/carla/issues/1561
I also faced the problem I can only control the car by W and S. YOU can change the statement auto as True then the car will move automatically.  BTW, I failed to move the car from A to B. Collision occurs

guochun7648, 2019-05-07:
> #1561
> I also faced the problem I can only control the car by W and S. YOU can change the statement auto as True then the car will move automatically. BTW, I failed to move the car from A to B. Collision occurs

Thank you for replying back. Yes, the car automatically moved. However, it did not work as expected from A to B. BTW, did I set it correctly for the coordinates provided above? If I did it in a wrong way, please let me know.

Thank you.

Derekabc, 2019-05-07:
i think it's enough for you to set the destination by `agent.set_destination([5.6, -88.9, 2.5])`

guochun7648, 2019-05-07:
> i think it's enough for you to set the destination by `agent.set_destination([5.6, -88.9, 2.5])`

Yes, I did to set up the destination in this way, but the car did not stop after it reached this point. It keep driving to somewhere else and seems like never stop. 
Also, after I use command spawn_npc.py added 100 cars in the same map. The car keeps shaking right and left instead of driving straight directly.
Is there any solution to that?

Thank you.

Derekabc, 2019-05-07:
Correct, The car cannot stop at the destination

closed 2019-07-13 by stale[bot]

bijayaniSMishra, 2021-04-21:
@guochun7648 and @Derekabc Facing the same issue. Did you guys found any solution for this?