carla-simulator/carla #1614: Retrieving Measurements from CARLA 0.9.5

arunk786, 2019-05-07:
I am using compiled version of Carla 0.9.5 and it all the examples are working fine. I have gone through the [https://carla.readthedocs.io/en/latest/measurements/](url). I am unable to retrieve all the measurements mentioned.  
For example, i could get the location and velocity and compute few more measurements like speed from velocity. Also, i am able to get some measurements like collision impulse from Collision sensor class modification.
But, i see that the following elegant way is not valid anymore to retrieve other measurements:
measurements, sensor_data = carla_client.read_data()
I am interested to retrieve the "collision" and "intersection" measurements and i would like to know if there is any way to do now.

arunk786, 2019-05-09:
Hi @nsubiron,

Thank you. Latest Docs are showing 0.9.x now. I have checked "Cameras and Sensors" section and i am using it.
Please clarify the below queries based on my understanding:
1). After seeing the APIs, a lot of earlier measurements like "intersection_otherlane /sidewalk " are not available now, with new system. But information that lane invasion event has occurred and for which actor is available now. 
Is there a way to get (a) current lane and neighboring lane information of the actor (b) actors on neighbor lane ?
2). Is there a way to access the "current high-level command provided by the topological planner" to the actor ?
3). I know we can record the 3d images in the simulator and also we can use the start recording and stop recording from programs and dump to file. I am interested in 2D real time images.
But, Is there any way to record 2d images of "no_rendering.py" ?
4). I am interested to build a simple map without any intersections or traffic lights and straight roads with two lanes and use it. (a) Is there a simpler way than "RoadRunner -> UnrealEngine -> Carla" ?  (b) Even if i use RoadRunner and the process can i use the map with my compiled version of CARLA 0.9.5 ?

nsubiron, 2019-05-09:
Hi @arunk786,

1) The road info can be retrieved now with our [Waypoint API](https://carla.readthedocs.io/en/latest/python_api_tutorial/#map-and-waypoints). You can retrieve a [waypoint](https://carla.readthedocs.io/en/latest/python_api/#carlawaypoint) from the vehicle location, the waypoint has information on lane id, road id, etc, as extracted from the OpenDrive file associated with the map.

2) If you're using the Python-side autopilot (automatic_control.py) I believe you can, but I'm not very familiar with that code. If you're using the server-side autopilot (default one), then unfortunately no, the route is chosen at random at every intersection.

3) The no-rendering mode script does not have the functionality for saving images to disk. However, the map image is internally stored as a pygame surface, I believe it would be relatively easy to modify the script to dump this surface to disk.

4) You could build your map with any software that allows exporting to Unreal Engine actually, but you'll need to provide an OpenDrive (XODR) file if you want the autopilot, waypoints API, and these things to work. We recommend RoadRunner because it automatically generates OpenDrive files and its very well tested and integrated with our pipeline. Either case, you can then compile a Carla release with the map included, or use [our tools](https://carla.readthedocs.io/en/latest/export_import_dist/) to create a "map package" that can be imported into a compiled version of Carla.

arunk786, 2019-05-09:
Your comments were really helpful. Thank you @nsubiron.

closed 2019-07-15 by stale[bot]