carla-simulator/carla #1615: Carla Agents Autopilot

Abanoub-G, 2019-05-07:
As I understand, there are two autopilots in CARLA for agents: 

1) The first one is the one that follows the route planner splines and is used in the spawn_npc.py example. I am trying to understand how does it follow the splines so I can apply changes etc. and hence trying to locate its code but I can't really locate where that is...? Can someone please help with locating these files and where the AI is so I can modify it if I wish?

2) The second AI is the one that doesn't need the splines and is used in automatic_control.py example. This one is clearly still in the development phase since it doesn't do much when it is run. This one's AI code is located in carla/PythonAPI/carla/agents/navigation/basic_agent.py and I am not interested in this one. 

Thanks

Abanoub-G, 2019-05-08:
Hi @nsubiron @felipecode,
I have asked this question a couple of times on different previous issues but no one seems to have the answer for it. I am solely held back by the issue of not being able to locate the AI code used in spawn_npc.py. Can please someone from Carla's team help with this? 

Thanks in advance

nsubiron, 2019-05-08:
Hi @Abanoub-G,

The autopilot code is in [WheeledVehicleAIController.cpp](https://github.com/carla-simulator/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/WheeledVehicleAIController.cpp), the code related to the route planners in [RoutePlanner.cpp](https://github.com/carla-simulator/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Traffic/RoutePlanner.cpp).

Abanoub-G, 2019-05-08:
Hi @nsubiron 
Thanks a lot for your prompt reply. I have investigated the code very well now. I have noticed that the vehicle follows the route using the way points in the array TargetLocations. I am bit confused about how the waypoints from the routeplanners get stored in the TargetLocations. It would be really kind of you if you can pinpoint for me where this actually happens?  Hence I can replace it with what I am trying to do, explained further in the paragraph below.   

I need to change this to replace it by a number of waypoints that i will supply from my python script. So that instead of the vehicles following the route planners they will follow the waypoint coordinates I will supply from my python script. If you already have a function like this that I am not aware of please do enlighten me. 

Please do not hesitate to mention any other pieces of info that you think I should know about in order to achieve my goal.

Thanks a lot.

nsubiron, 2019-05-09:
Hi @Abanoub-G,

The routes are assigned by the RoutePlanner, when a vehicle enters the trigger box of a RoutePlanner, `ARoutePlanner::OnTriggerBeginOverlap` is called, then a random route is chosen and assigned to the vehicle's AI controller. You could disable this functionality here and call `AWheeledVehicleAIController::SetFixedRoute` directly from your code.

To make sure your changes are compiled you can use `make launch` to launch the editor, this compiles any files that have changed.

Abanoub-G, 2019-05-10:
Hi @nsubiron ,
Thanks a lot for pointing out the AWheeledVehicleAIController::SetFixedRoute . However, I can't figure out how to use it. There is two things from my understanding that I need to do in order to make the vehicle follow my route. Please do correct me if I am wrong:

1) Pass my list of waypoints from my python script to the C++ WheeledVehicleAIController.cpp script. I have no clue how to do that. Is there a dedicated function that you guys have for passing from python to C++?

2) Then use the SetFixedRoute function to assign these waypoints to the vehicle; but I am not sure also where to call this function in the   WheeledVehicleAIController.cpp script in order to assign this waypoints to the vehicle. For example should it be called in the AWheeledVehicleAIController::TickAutopilotController() ?

Thanks

nsubiron, 2019-05-10:
Hi @Abanoub-G,

Calling a method from Python is quite involved as you need to make changes in the whole pipeline from Python to UE4. I would recommend taking a look at how "set_autopilot" method is implemented as is sort of similar.

1. You create a new RPC binding in CarlaServer, like [set_actor_autopilot](https://github.com/carla-simulator/carla/blob/9fde8f20cdd4301833ab7fc81764e5c8ca8e34f3/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Server/CarlaServer.cpp#L563-L585).
2. In the client side, you create a new method to calling this binding, [Client::SetActorAutopilot](https://github.com/carla-simulator/carla/blob/9fde8f20cdd4301833ab7fc81764e5c8ca8e34f3/LibCarla/source/carla/client/detail/Client.cpp#L231-L233).
3. Wrap it in the Simulator class, [Simulator::SetVehicleAutopilot](https://github.com/carla-simulator/carla/blob/9fde8f20cdd4301833ab7fc81764e5c8ca8e34f3/LibCarla/source/carla/client/detail/Simulator.h#L251-L253).
4. Create a method in the Vehicle class, [Vehicle::SetAutopilot](https://github.com/carla-simulator/carla/blob/9fde8f20cdd4301833ab7fc81764e5c8ca8e34f3/LibCarla/source/carla/client/Vehicle.cpp#L31-L33).
5. Finally, bind that method with Boost.Python, [vehicle.set_autopilot](https://github.com/carla-simulator/carla/blob/9fde8f20cdd4301833ab7fc81764e5c8ca8e34f3/PythonAPI/carla/source/libcarla/Actor.cpp#L92).

You don't need to modify the controller, by setting a fixed route the autopilot will already follow those waypoints.

Abanoub-G, 2019-05-10:
Hi @nsubiron 
Thanks a lot for your reply. Will try to implement this and get back to you with any questions; but before I get indulged with this process I would like to double check on two points:

1) How would you use the AWheeledVehicleAIController::SetFixedRoute function normally to set a route? As in what were you proposing how I use it when you mentioned it to me?

2) Also, just to double check there isn't a way where you can tell a vehicle to go to a location from the python scripts that already exists. For example, you can't say GoToLocation(carla.location(x), carla.location(y), carla.location(z)) on the python script and the car will do it, yeah? Note: I don't mean navigate to this location I mean just go straight ahead to it.

Thanks

nsubiron, 2019-05-13:
1. Just provide the waypoints you want to this function and the vehicle will follow them.
2. No, unfortunately there is no method in our API to tell a vehicle to go to a location.

Derekabc, 2019-05-27:
@Abanoub-G Any update to your problem?  Hope you can share your experience about how to let the car follow predetermined route!  Thanks

closed 2019-08-02 by stale[bot]

linxigjs, 2019-09-30:
> Hi @Abanoub-G,
>
> Calling a method from Python is quite involved as you need to make changes in the whole pipeline from Python to UE4. I would recommend taking a look at how "set_autopilot" method is [...]

> Hi @Abanoub-G,
>
> Calling a method from Python is quite involved as you need to make changes in the whole pipeline from Python to UE4. I would recommend taking a look at how "set_autopilot" method is [...]

thx @nsubiron, i tryed and succeeded in adding a new PythonAPI function as your suggested pipeline, with input parameter is a single float number. But how can i use a python list parameter sending to C++ Server side and translating to a std::vector?

AravindChandradoss, 2019-10-01:
I am looking for this functionality. @linxigjs @Abanoub-G any update? where you able to implement those functionalities? 

@nsubiron Is this the only way to send/receive data (like waypoints) from python API to server? Coz, my need is like "I want to collect data (waypoints from the server with timestamp), so that I can calculate features like distance/orientation to next_waypoint, distance to lane center and other such feature" I know I can use get_waypoint() method and use basic_agent or roaming_agent -- on client-side, however, I want to utilize the Autopilot functionality (Server-side). Any suggestion/direction would be really helpful. Thanks in advance.

stavBodik, 2019-12-11:
Please look at the file which actually should install the dist folder, Util\BuildTools\BuildPythonAPI.bat  in my case this like had bug : py -3 setup.py bdist_egg . . . remove -3 and run again.