carla-simulator/carla #1617: how to modify tagger files in carla 0.8.4

geekhut, 2019-05-08:
I want to include every class of objects in my segmentation images to get my bounding box drawn. How to modify the Tagger files to achieve this?

geekhut, 2019-05-08:
I did add more tags but only some of the tags worked, and tags for pedestrians don't work at all.. do you have the code example? thank you. 

> You can add more tags to Tagger.cpp and extend the functions there to identify more classes.

nsubiron, 2019-05-08:
These are the methods that determine the tag based on the mesh path, the mesh needs to be stored in a folder with the tag name (actually hard-coded to 3rd level folder must be tag name).

https://github.com/carla-simulator/carla/blob/ab9a2c7cb524d279687e4f56228cb6244277a229/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/Tagger.cpp#L47-L70

geekhut, 2019-05-09:
> These are the methods that determine the tag based on the mesh path, the mesh needs to be stored in a folder with the tag name (actually hard-coded to 3rd level folder must be tag name). [...]

well I did this to include all sorts of pedestrians:
`  //---------------------pedestrains----------------------------------------------
  //thin
  else if (String == "Female_Dress_Shoes_Thin_rig")                        return ECityObjectLabel::Female_Dress_Shoes_Thin_rig;
  else if (String == "Female_SkinnyJeans_Tshirt_Boots_Thin_rig")           return ECityObjectLabel::Female_SkinnyJeans_Tshirt_Boots_Thin_rig;
  else if (String == "Female_Skirt_Shirt_SBoots_thin_rig")                 return ECityObjectLabel::Female_Skirt_Shirt_SBoots_thin_rig;
  else if (String == "Male_Shorts_Polo_Shoes_thin_rig")                    return ECityObjectLabel::Male_Shorts_Polo_Shoes_thin_rig;
  else if (String == "Male_SkinnyJeans_Coat_Shoes_thin_rig")               return ECityObjectLabel::Male_SkinnyJeans_Coat_Shoes_thin_rig;
  else if (String == "Male_SkinnyJeans_TShirt_Shoes_thin_rig")             return ECityObjectLabel::Male_SkinnyJeans_TShirt_Shoes_thin_rig;
  //small
  else if (String == "Female_Dress_Shoes_Small_rig")                       return ECityObjectLabel::Female_Dress_Shoes_Small_rig;
  else if (String == "Female_SkinnyJeans_Tshirt_Boots_small_rig")          return ECityObjectLabel::Female_SkinnyJeans_Tshirt_Boots_small_rig;
  else if (String == "Female_Skirt_Shirt_SBoots_Small_rig")                return ECityObjectLabel::Female_Skirt_Shirt_SBoots_Small_rig;
  else if (String == "Male_Shorts_Polo_Shoes_small_rig")                   return ECityObjectLabel::Male_Shorts_Polo_Shoes_small_rig;
  else if (String == "Male_SkinnyJeans_Coat_small_rig")                    return ECityObjectLabel::Male_SkinnyJeans_Coat_small_rig;
  else if (String == "Male_SkinnyJeans_TShirt_Shoes_small_rig")            return ECityObjectLabel::Male_SkinnyJeans_TShirt_Shoes_small_rig;
  //base
  else if (String == "Female_Dress_Shoes")                                 return ECityObjectLabel::Female_Dress_Shoes;
  else if (String == "Female_SkinnyJeans_Tshirt_Boots")                    return ECityObjectLabel::Female_SkinnyJeans_Tshirt_Boots;
  else if (String == "Female_Skirt_Shirt_SBoots")                          return ECityObjectLabel::Female_Skirt_Shirt_SBoots;
  else if (String == "Male_Shorts_Polo_Shoes")                             return ECityObjectLabel::Male_Shorts_Polo_Shoes;
  else if (String == "Male_SkinnyJeans_Coat_Shoes")                        return ECityObjectLabel::Male_SkinnyJeans_Coat_Shoes;
  else if (String == "Male_SkinnyJeans_TShirt_Shoes_Rig")                  return ECityObjectLabel::Male_SkinnyJeans_TShirt_Shoes_Rig;`

geekhut, 2019-05-09:
> These are the methods that determine the tag based on the mesh path, the mesh needs to be stored in a folder with the tag name (actually hard-coded to 3rd level folder must be tag name). [...]

Hi, what does this segment of code do? How do I modify it if I were to add more tags like:
 
`else if (String == "Female_Dress_Shoes_Thin_rig")                      
       return ECityObjectLabel::Female_Dress_Shoes_Thin_rig;`
`else if (String == "DodgeChargePolice")           
       return ECityObjectLabel::DodgeChargePolice;`

to Tagger.cpp script? thank you!

nsubiron, 2019-05-09:
Static meshes are stored under `Content/Carla/Static/TagName/...`, this method is looking to see if the TagName folder matches any of the tags. You can extend this methods to use different folder names or maybe subfolders.

geekhut, 2019-05-10:
> Static meshes are stored under `Content/Carla/Static/TagName/...`, this method is looking to see if the TagName folder matches any of the tags. You can extend this methods to use different folder names or maybe subfolders.

I have done all the tagging according to TagName folder but still, only some of the objects get tagged, I am not sure how to edit this segment of code when I add more tags in Tagger.cpp

_template
static ECityObjectLabel GetLabelByPath(const T *Object)
{
const FString Path = Object->GetPathName();
TArray StringArray;
Path.ParseIntoArray(StringArray, TEXT("/"), false);
return (StringArray.Num() > 3 ? GetLabelByFolderName(StringArray[3]) : ECityObjectLabel::None);
}_

closed 2019-07-16 by stale[bot]