carla-simulator/carla #1623: Problem with running manual_control.py in Carla 0.9.4

Suryavf, 2019-05-08:
Hello

I am new to Carla. I'm following the tutorial:
[https://www.youtube.com/watch?v=AaJekfFR1KQ&t=186s](https://www.youtube.com/watch?v=AaJekfFR1KQ&t=186s)

I ran without problems:
`./CarlaUE4.sh HDMaps/Town01 -carla-server -benchmark -fps=15 -windowed -ResX=800 -ResY=600`
However, the simulation was without the car. Only view from camera.

Next, I executed:
`python PythonAPI/examples/manual_control.py`
Resulting:
`Traceback (most recent call last):`
`  File "PythonAPI/examples/manual_control.py", line 72, in <module>`
`    from carla import ColorConverter as cc`
`ImportError: cannot import name ColorConverter`

What am I doing wrong?
Is it wrong to not see the car when executing `./CarlaUE4.sh`?

Thanks!

nsubiron, 2019-05-09:
Hi @Suryavf,

That tutorial refers to 0.8.x versions, quite different from latest 0.9.x. In newer versions, the simulator window does not follow the vehicle, it displays a flying camera (spectator), this was done like this because in newer versions there is no longer a concept of player as any vehicle can be controlled. Check out our [Gettings started](https://carla.readthedocs.io/en/latest/getting_started/), that was updated more recently.

As for the import error, you probably have installed an older version of the carla Python module `pip uninstall carla` to make sure you don't have it installed.

One final comment, `HDMaps/Town01` is not going to work, use simply `Town01`. HD Maps are pcd files, not loadable by the simulator.

rgourova, 2019-05-09:
Hi @Suryavf,

To add to nsubiron comment regarding the import error, you should start the manual_control.py file from its folder (python manual_control.py) or adjust the script accordingly to load correctly the carla module from the correct path. If you look at the code it goes two folders back ('../carla/dist/') to look for the carla module and as you are not in the PythonAPI/examples folder it goes to some other folders and it does not find the module. 

Regarding your question whether you should see any cars when you execute ./CarlaUE4.sh - no, you should not see any cars, you are just starting the server and you should populate it with actors by executing other scripts (spawn_npc.py for example).

Suryavf, 2019-05-09:
Thank you very much for your answers!

@nsubiron I tried `pip uninstall carla`. I checked that I do not have any version of carla installed.
@rgourova Now I executed the code from `PythonAPI/examples` folder. I ran `python spawn_npc.py` resulting in:
`done.`
`Traceback (most recent call last):`
`  File "spawn_npc.py", line 124, in <module>`
`    main()`
`  File "spawn_npc.py", line 65, in main`
`    client = carla.Client(args.host, args.port)`
`AttributeError: 'module' object has no attribute 'Client'`

What am I doing wrong?

rudrapoudel, 2019-06-12:
Hi,

make sure you are using either 

> python 2.7 or 3.5

 because they have compiled code for those two versions only. Look at the directory PythonAPI/carla/dist from complied distribution, it has following files only, 

> carla-0.9.5-py2.7-linux-x86_64.egg
> carla-0.9.5-py3.5-linux-x86_64.egg
> 

And scripts add lib into sys.path based on your python's major and minor only!

```
try:
    sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass
```

The relevant codes are here (in the source from github) in the C++,

> carla-0.9.5/PythonAPI/carla/source/libcarla => Client.cpp

Hope this help!

closed 2019-08-18 by stale[bot]

itserror404, 2023-07-24:
hi, were you able to fix this?