carla-simulator/carla #1625: 0.9.5 and python3

vishal-prgmr, 2019-05-09:
OS: Win 7
UE: 4.21.2
Python: 3.7.3
Carla: 0.9.5

Issue: Running make PythonAPI inside an environment which has python3 in it,  throws the following error 

 -Make sure "py" is installed.
 -Make sure it is available on your Windows "py".

huuanhhuynguyen, 2019-05-09:
Vishal, I think the most stable version is 0.8.2

closed 2019-05-20 by vishal-prgmr

krish97git, 2019-09-26:
Yes, Unable to run any py scripts
Its says  load_dynamic
    return _load(spec)
ImportError: DLL load failed: %1 is not a valid Win32 application.
any specific reason behind this? Many people suggested 32bit-64bit version problem. But unable to solve this.

vishal-prgmr, 2019-09-26:
@krish97git, my problem was a mix of python versions. I cleaned all the existing versions and installed python3 alone and then built Carla from within a conda environment.

krish97git, 2019-09-26:
@vishal-prgmr 
Is that working fine? 
And are all the python API interfacing well?