carla-simulator/carla #1626: Cannot import ColorConverter [0.9.5]

huuanhhuynguyen, 2019-05-09:
When I run `manual_control.py`, I got the importation error:
```
Traceback (most recent call last):
  File "/home/ngh7rng/carla/CARLA_0.9.5/PythonAPI/examples/manual_control.py", line 72, in <module>
    from carla import ColorConverter as cc
ImportError: cannot import name 'ColorConverter'
```
I already added PythonAPI and carla as roots to PYTHONPATH.
I could also not able to find out manually where is actually the class `ColorConverter` in the `carla` folder.

huuanhhuynguyen, 2019-05-09:
Ah I see. There are Python egg files in the PythonAPI/carla/dist/.
Just need to run terminal command:
```
easy_install ./carla-0.9.5-py3.5-linux-x86_64.egg
```

closed 2019-05-09 by huuanhhuynguyen

yasser-h-khalil, 2020-07-22:
Hello,

I am facing the same issue. I tried easy_install but I get permission denied. I tried that statement also with "sudo", still didnt work.
I followed the carla build tutorial for linux line by line. But the last step didnt spawn the vehicles. It said "ImportError: cannot import name VehicleLightState"

Your help is much appreciated.

sdhanase-valeo, 2020-07-23:
Hi Yasser,

In the instructions they asked to use python3.. I think this is related something with how python2 and python3 are handling the  dependency issues.. So, after successful $ make PythonAPI.. Goto examples folder - $ cd PythonAPI/examples and run $python spawn_npc.py

yasser-h-khalil, 2020-07-23:
Hi, 

Thanks for your response. 
Yes, it worked with python2.7.
I noticed Carla installed python2.7 and 3.6. With python3.6, running spawn_npc.py gives import error. The same type of error appears when executing python3 manual_control.py.

vinayakap, 2021-09-03:
> 
> 
> Ah I see. There are Python egg files in the PythonAPI/carla/dist/.
> Just need to run terminal command:
> 
> ```
> easy_install ./carla-0.9.5-py3.5-linux-x86_64.egg
> ```

Thanks for highlighting.  This is nowhere is documented.