carla-simulator/carla #1627: Error: Failed to link program in carla 0.9.4

wangyuanbiubiubiu, 2019-05-10:
When I run Carla 0.9.4 on Linux 16.0.4

`    [2019.05.10-01.26.33:099][  0]LogRHI: Error: Failed to link program. Current total programs: 131 binary bytes: 402548159
    log:
    error: vertex shader output 'out_TEXCOORD0' specifies noperspective interpolation qualifier, but fragment shader input specifies no interpolation qualifier

    Failed to find symbol file, expected location:
    "/home/wya/CARLA_0.9.4/CarlaUE4/Binaries/Linux/CarlaUE4.sym"
    Fatal error: [File:/var/lib/jenkins/UnrealEngine_4.21/Engine/Source/Runtime/OpenGLDrv/Private/OpenGLShaders.cpp] [Line: 3718]
    Failed to link program. Current total programs: 131, precompile: 0
    Signal 11 caught.
            Malloc Size=65538 LargeMemoryPoolOffset=65554`

nsubiron, 2019-05-10:
It seems like same issue: [Shader compilation crash with AMD GPU](https://answers.unrealengine.com/questions/733926/shader-compilation-crash-with-amd-gpu.html).

pawel-ziecina, 2019-06-11:
It's similar to #1210 and also observable on 0.9.5 release

```
[2019.06.11-09.31.59:118][  0]LogRHI: Error: Failed to link program. Current total programs: 134 binary bytes: 409286582
  log:
error: vertex shader output `out_TEXCOORD0' specifies noperspective interpolation qualifier, but fragment shader input specifies no interpolation qualifier

Failed to find symbol file, expected location:
"/tmp/carla-20190400-095-original/CarlaUE4/Binaries/Linux/CarlaUE4.sym"
Fatal error: [File:/var/lib/jenkins/UnrealEngine_4.21/Engine/Source/Runtime/OpenGLDrv/Private/OpenGLShaders.cpp] [Line: 3718] 
Failed to link program. Current total programs: 134, precompile: 0
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=68112 LargeMemoryPoolOffset=199248 
Malloc Size=128368 LargeMemoryPoolOffset=327632 
Malloc Size=501903 LargeMemoryPoolOffset=829551 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```

- Ubuntu 16.04 and 18.04 on 3 laptops with intel GPU
- observable also while using -opengl3 flag
- glxinfo -B output from ubuntu 16.04 laptop:

```
name of display: :0
display: :0  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Intel Open Source Technology Center (0x8086)
    Device: Mesa DRI Intel(R) HD Graphics 630 (Kaby Lake GT2)  (0x591b)
    Version: 18.0.5
    Accelerated: yes
    Video memory: 3072MB
    Unified memory: yes
    Preferred profile: core (0x1)
    Max core profile version: 4.5
    Max compat profile version: 3.0
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) HD Graphics 630 (Kaby Lake GT2) 
OpenGL core profile version string: 4.5 (Core Profile) Mesa 18.0.5
OpenGL core profile shading language version string: 4.50
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile

OpenGL version string: 3.0 Mesa 18.0.5
OpenGL shading language version string: 1.30
OpenGL context flags: (none)

OpenGL ES profile version string: OpenGL ES 3.2 Mesa 18.0.5
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
```

pawel-ziecina, 2019-06-11:
output of `glxinfo -B` for laptop with Ubuntu 18.04

```             
name of display: :0
display: :0  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Intel Open Source Technology Center (0x8086)
    Device: Mesa DRI Intel(R) HD Graphics 630 (Kaby Lake GT2)  (0x591b)
    Version: 18.2.8
    Accelerated: yes
    Video memory: 3072MB
    Unified memory: yes
    Preferred profile: core (0x1)
    Max core profile version: 4.5
    Max compat profile version: 3.0
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) HD Graphics 630 (Kaby Lake GT2) 
OpenGL core profile version string: 4.5 (Core Profile) Mesa 18.2.8
OpenGL core profile shading language version string: 4.50
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile

OpenGL version string: 3.0 Mesa 18.2.8
OpenGL shading language version string: 1.30
OpenGL context flags: (none)

OpenGL ES profile version string: OpenGL ES 3.2 Mesa 18.2.8
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
```

pawel-ziecina, 2019-06-11:
also confirmed on more recent builds (from carla jenkins server):

nsubiron, 2019-06-11:
Try running the last version with Vulkan instead of OpenGL. Install Vulkan with

    sudo apt-get install libvulkan1

then run Carla normally, it should default to Vulkan. Note though that Vulkan doesn't work without display.

Derekabc, 2019-06-20:
@nsubiron how to run the carla with Vulkan? I installed libvulkan, but the problem still exist.

pawel-ziecina, 2019-06-21:
I've to install also `mesa-vulkan-drivers vulkan-utils`. 

I've used laptop with ubuntu 16.04, integrated intel GPU and CARLA jenkins build based on 649699b commit, but it still failing.

```
[2019.06.21-07.56.19:278][  0]LogLoad: Game class is 'CarlaGameMode_C'
[2019.06.21-07.56.19:359][  0]LogWorld: Bringing World /Game/Carla/Maps/Town03.Town03 up for play (max tick rate 0) at 2019.06.21-09.56.19
[2019.06.21-07.56.19:360][  0]LogCarlaServer: Initialized CarlaServer: Ports(rpc=2000, streaming=2001)
[2019.06.21-07.56.19:374][  0]LogWorld: Bringing up level for play took: 0.093486
[2019.06.21-07.56.19:387][  0]LogCarlaServer: New episode 'Town03' started
[2019.06.21-07.56.19:387][  0]LogLoad: Took 2.911791 seconds to LoadMap(/Game/Carla/Maps/Town03)
[2019.06.21-07.56.19:586][  0]LogLoad: (Engine Initialization) Total time: 4.40 seconds
[2019.06.21-07.56.19:646][  0]LogRenderer: Reallocating scene render targets to support 1600x900 Format 10 NumSamples 1 (Frame:1).
[2019.06.21-07.56.19:687][  0]LogRenderer: Warning: Gaussian DOF algorithm is deprecated and will be removed from deferred shading renderer. Consider using the Circle DOF method in post process settings.
CommonUnixCrashHandler: Signal=11
[2019.06.21-07.56.27:492][  0]LogCore: Error: appError called: Assertion failed: bSuccess [File:/var/lib/jenkins/UnrealEngine_4.22/Engine/Source/Runtime/VulkanRHI/Private/VulkanCommandBuffer.cpp] [Line: 364] 

[2019.06.21-07.56.27:502][  0]LogCore: === Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to read memory at address 0x00000000000017bf

[2019.06.21-07.56.27:502][  0]LogCore: Assertion failed: bSuccess [File:/var/lib/jenkins/UnrealEngine_4.22/Engine/Source/Runtime/VulkanRHI/Private/VulkanCommandBuffer.cpp] [Line: 364] 

0x0000000004165311 CarlaUE4!GameThreadWaitForTask(TRefCountPtr<FGraphEvent> const&, ENamedThreads::Type, bool) [/var/lib/jenkins/UnrealEngine_4.22/Engine/Source/Runtime/RenderCore/Private/RenderingThread.cpp:1091]
0x000000000416130a CarlaUE4!FlushRenderingCommands(bool) [/var/lib/jenkins/UnrealEngine_4.22/Engine/Source/Runtime/RenderCore/Private/RenderingThread.cpp:1225]
0x0000000004471154 CarlaUE4!FSlateRHIRenderer::FlushCommands() const [/var/lib/jenkins/UnrealEngine_4.22/Engine/Source/Runtime/SlateRHIRenderer/Private/SlateRHIRenderer.cpp:1273]
0x0000000002fe730c CarlaUE4!FLinuxApplication::ProcessDeferredMessage(SDL_Event) [/var/lib/jenkins/UnrealEngine_4.22/Engine/Source/Runtime/ApplicationCore/Private/Linux/LinuxApplication.cpp:699]
0x0000000002fe55df CarlaUE4!FLinuxPlatformApplicationMisc::PumpMessages(bool) [/var/lib/jenkins/UnrealEngine_4.22/Engine/Source/Runtime/ApplicationCore/Private/Linux/LinuxPlatformApplicationMisc.cpp:415]
0x00000000024a6649 CarlaUE4!FEngineLoop::Tick() [/var/lib/jenkins/UnrealEngine_4.22/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:3882]
0x00000000024adf5a CarlaUE4!GuardedMain(char16_t const*) [/var/lib/jenkins/UnrealEngine_4.22/Engine/Source/Runtime/Launch/Private/Launch.cpp:170]
0x0000000005bdd70f CarlaUE4!CommonUnixMain(int, char**, int (*)(char16_t const*)) [/var/lib/jenkins/UnrealEngine_4.22/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:243]
0x00007f17343ca830 libc.so.6!__libc_start_main(+0xef)
0x000000000249d029 CarlaUE4!_start()

[2019.06.21-07.56.27:516][  0]LogExit: Executing StaticShutdownAfterError
[2019.06.21-07.56.27:528][  0]LogCore: Warning: Unable to statfs('/home/pawel/sandbox/carla/releases/CARLA_649699b/CarlaUE4/Saved/Crashes/crashinfo-CarlaUE4-pid-6079-0BD2871D08D6F62EBA5E1942BB5DC87C/CarlaUE4.log'): errno=2 (No such file or directory)
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
```
[vulkan info](https://github.com/carla-simulator/carla/files/3313532/vulkan.txt)
[full_carla.log](https://github.com/carla-simulator/carla/files/3313534/carla.log)

pawel-ziecina, 2019-08-02:
Newest Carla 0.9.6 release with -opengl flag works ok on laptop with Intel GPU

iamycee, 2019-09-01:
Same. Why are the PythonAPI examples different for 0.9.5 though?

using 0.9.6 with 
` client = carla.Client('localhost', 2000) `
gives an error "module has no attribute Client"

in 0.9.6, the examples use the ` make_carla_client ` syntax. 

Why are there these inconsistencies or am I missing something here?

athkul98, 2019-10-31:
I have tried sudo apt-get install libvulkan1
and
sudo apt-get install mesa-vulkan-drivers vulkan-utils
and it still doesn't work. Still gives the same error. I have an Intel GPU running Carla release 0.9.4 on Linux Ubuntu 16.04 LTS laptop.

closed 2020-01-06 by stale[bot]