carla-simulator/carla #1629: make PythonAPI - Waypoint.cpp build error

vishal-prgmr, 2019-05-10:
Running make PythonAPI throws an error while building Waypoint.cpp. 
Attaching a screenshot of the error. 
[image]

closed 2019-05-14 by vishal-prgmr

portends, 2019-05-23:
Hi @vishal-prgmr 

I am getting the same error for waypoint.cpp when making PythonAPI. Have you found a method to resolve the error?

vishal-prgmr, 2019-05-23:
@portends , uninstall all the python versions you have on your local machine. There should not be any conflict between 2.7 and 3.7. 

Install a fresh copy of Anaconda with python 3.  Run **make PythonAPI** from within an environment with all the prerequisites installed (numpy, pygame blah blah which you can find in the installation docs)