carla-simulator/carla #1630: How can I get recorder collisions images/videos

morningsky, 2019-05-10:
According by [python_api_tutorial](https://carla.readthedocs.io/en/latest/python_api_tutorial/), we can collisions  info by `client.replay_file("col2.log", 13, 0, 122)`, and it will print info about collisions (I use jupyter notebook), but how can I get the collision image or video, such as
[image]

bernatx, 2019-05-13:
Following the tutorial the steps are:

- Find the time that you are interested in. For that you can use the 'client.show_recorder_actors_blocked()' API to find when the vehicle got blocked, for example.

- Then with that time you can start the replayer at that exact moment and capture the images you need from the simulation, using any screen capture or similar software. 

That is how it was done in the tutorial.

morningsky, 2019-05-13:
@bernatx  Thanks！ But... How can I start the replayer at that exact moment "? Which API can do this or command?

bernatx, 2019-05-13:
The command you wrote from the tutorial ("client.replay_file("col2.log", 13, 0, 122)") means to start replaying the file 'col2.log' at time 13 (seconds), until the end of the replay, and following the vehicle with id.122. So, the second parameter is the time to start replaying.

morningsky, 2019-05-15:
@bernatx Thanks very much! I know the mean of these params, but how can I get the images/vews when cilent.repaly_file()? when I run the code client.replay_file, it just prints info on terminal, and the pygame windows keep no change...

bernatx, 2019-05-15:
Which info do you get from terminal?
It should say it is replaying the file, and then the server starts replaying the simulation.

morningsky, 2019-05-15:
@bernatx  Thanks, I solved it. When I run CarlaUE4 it start a window 1,  then I run manual_control.py it start  window 2. when I run the code  client.replay_file , the replaying video running in the window 1.... I just focus on windows 2.
And I have another problem, when I run CarlaUE4, my mouse blocked in the window 1, I have to get the mouse point by use Win+D, it's not convenient， Do you have any advice？
[image]

closed 2019-05-17 by morningsky