carla-simulator/carla #1632: ERROR: Invalid session: no stream available with id ...

nonsingularmatrix, 2019-05-11:
Hello,

I am using Carla 0.9.5 and sometimes getting the error message **"ERROR: Invalid session: no stream available with id ..."**. Does anyone know possible causes for this error?

In my use case several vehicles are getting spawned. Hereby, I always attach collision and obstacle sensors to them. In case the collision sensor gets triggered during a simulation run, I reset the whole simulation (stopping and destroying the attached sensors, destroying vehicles)  and repeat the whole process in a similar way.

Thanks for your help.

slinshady89, 2019-05-11:
Hello,
I got the same issue just running Carla 0.9.5 on Ubuntu 16.04 with the following command at startup:
./CarlaUE4.sh -carla-server -windowed -ResX=320 -ResY=240 -benchmark -fps 10
and I have problems to connect the carla_ros_bridge_with_example_ego_vehicle to the server which worked fine until now.
Also that isn't connecting any more because of this issue (maybe related)
File "/home/nils/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/ego_vehicle.py", line 115, in send_vehicle_msgs
    vehicle_info.id = self.carla_actor.id
AttributeError: 'CarlaEgoVehicleInfo' object has no attribute 'id'

nonsingularmatrix, 2019-05-12:
I could finally fix the problem in my specific use case.
In some simulation runs I did not delete all created sensors before restarting with a new simulation run.

slinshady89, 2019-05-13:
For me the same apparently. I saw a high memory usage in top by a carla_ros_bridge process while none should run. To kill this helped me.
I think you can close this @nonsingularmatrix

closed 2019-05-13 by nonsingularmatrix

Yumin-Sun-00, 2020-04-12:
I am using 0.9.8

I got the same problem when I do ` CUDA_VISIBLE_DEVICES=1 DISPLAY= ./CarlaUE4.sh -quality-level=Epic -opengl`

Yumin-Sun-00, 2020-04-12:
> I am using 0.9.8
> 
> I got the same problem when I do ` CUDA_VISIBLE_DEVICES=1 DISPLAY= ./CarlaUE4.sh -quality-level=Epic -opengl`

Solved. You just need to switch to another port, specify it by add `--carla-world-port=3000` at the end of `CUDA_VISIBLE_DEVICES=1 DISPLAY= ./CarlaUE4.sh -quality-level=Epic -opengl`

Horcrux1, 2020-08-30:
Setting `-carla-streaming-port=0` when running Carla simulator solved the problem for me

cinjon, 2021-03-30:
@Horcrux1  / @Yumin-Sun-00 , why would those two _solve_ the issue?

GooVincent, 2021-11-29:
> Solved. You just need to switch to another port, specify it by add `--carla-world-port=3000` at the end of `CUDA_VISIBLE_DEVICES=1 DISPLAY= ./CarlaUE4.sh -quality-level=Epic -opengl`

Please why the default port 2000 would prompt the 'invalid session: no stream avaiable...' issue

YuanYunshuang, 2022-11-04:
Just to make a note here of how I met this issue.
The sensors were destroyed inappropriately. For example, they were attached to some vehicles, I saw this issue when I destroyed the vehicle before the attached sensors (I thought theoretically the attached sensors would be destroyed along with the vehicle). When I first destroy all sensors and then vehicles, this issue disappeared.

AbdlRahman2020, 2023-02-17:
> 

Could you illustrate more on what you mean by destroy the sensors properly? 

For example I tried something like this but it didn't work.
`        self.sensors = []
        self.vehicle = None
`

Morphlng, 2023-04-22:
> For example I tried something like this but it didn't work. ` self.sensors = [] self.vehicle = None`

A `carla.Actor` object needs to call `destroy` method to properly destroy them. You have to loop through your sensors first and call destroy on each of it:

```python
for sensor in self.sensors:
  if sensor.is_listening:
    sensor.stop()
  sensor.destroy()

self.vehicle.destroy()
```