carla-simulator/carla #1633: Python3.7

kimiya66, 2019-05-11:
Hi,
how can I make Carla to work with python3.7. I updated carla to the last version but it is still not working with python3.7
because I need to connect to matlab. matlab.engine could just connected to python3.7 and not even 3.6 in my pc. 
when I try to read python files with python3.7, I got this error:
"No module named 'carla'" and it has problem with the line 'import carla' but for other python versions as python2.7 or python3.6 it realizes carla.
Anyone can help me with this.I would be thankful.

wangludong, 2019-05-13:
I think you can just fix the system path,replace it with yours for sys.path.append,this is what i did:
sys.path.append("../carla/dist/carla-0.9.5-py3.5-linux-x86_64.egg")

kimiya66, 2019-05-13:
> I think you can just fix the system path,replace it with yours for sys.path.append,this is what i did:
> sys.path.append("../carla/dist/carla-0.9.5-py3.5-linux-x86_64.egg")
> My python version is 3.6, and it works well with 3.5,just a temporary expedient.

Hi,
thanks for your answer. where should I make these changes?

kimiya66, 2019-05-13:
@wangludong Would you please telling me where should I change sys path? I would be grateful.

kimiya66, 2019-05-13:
> @wangludong Would you please telling me where should I change sys path? I would be grateful.

I add this sys.append line in my code(carla client) but still not working!

wangludong, 2019-05-14:
I changed it in PythonAPI/examples/manual_control.py, you may search for "sys.path.append".
I replaced following lines with mine:
sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
--------->
sys.path.append("../carla/dist/carla-0.9.5-py3.5-linux-x86_64.egg")

closed 2019-07-20 by stale[bot]