carla-simulator/carla #1635: How to avoid driving out of the map?

wangludong, 2019-05-13:
Hi
When i drive across the edge of the map,it seems like the car falls off,and that looks queer for my app.So i'm wondering if there is any solution.
Thank you very much.

wattsavolt, 2019-05-13:
You can disable gravity on the physics component of the vehicle in Unreal. This currently cannot be done through the Carla python API. You would have to modify the vehicles in Unreal or modify the LibCarla server side code to disable the gravity for the vehicle being controlled.

wangludong, 2019-05-14:
Thanks for your help.
I can find the gravity option in Unreal when i click a vehicle.But i need to generate a vehicle dynamically with a python script, and have no idea about how to control the gravity in this situation.Could you give me some more details?

p-schulz, 2019-05-14:
You may want to add some bounding volumes at the end of a map - something like cubes, and a plane underneath the map. 

Also try looking for the "Kill ZVolume": 
https://www.epicgames.com/unrealtournament/forums/unreal-tournament-development/ut-development-level-design/learning-level-design/8058-how-to-create-a-player-death-zone-when-the-player-falls-off-the-map-they-die

This could crash your python script, so be sure to catch any exceptions.

closed 2019-07-20 by stale[bot]