carla-simulator/carla #1636: How can we get the driver view?

morningsky, 2019-05-13:
When run carla in control manual mode, I know we can use tab to change the view. But how can we get the truth driver view? like that 
[image]

MaxJohnsen, 2019-05-14:
You either need to add a new camera, or change the transformation of one of the existing cameras.

Changing:
https://github.com/carla-simulator/carla/blob/cf58f7a2c1a57b344a0a033b8522823206c901c0/PythonAPI/examples/manual_control.py#L667

to:
```python
carla.Transform(carla.Location(x=-0.15,y=-0.4, z=1.2), carla.Rotation()),
```
will result in  the following view:

[image]

sheepwxy, 2019-05-15:
From some drive views I saw some cars are black inside, is this because the location of camera need to be modified or the car model is not perfect?

morningsky, 2019-05-15:
@sheepwxy  I met the same problem

morningsky, 2019-05-15:
@MaxJohnsen Thank you very much for your reply！ Follow your advice, I get the better view on Tesla, but for some other vehicle, I get the worst view, such as 
[image] How can I find the best params for them or it is limited by the simulation model?
[image]

analog-cbarber, 2019-05-15:
First, I don't think there is any guarantee that every car has a reasonable interior for this purpose. Second, every vehicle has different proportions, so you really need to look at the mesh in the editor to figure out the appropriate spot. For my project, I created a file that lists named external sensor mountpoints for the available vehicles., but I did not attempt to measure driver head position.

MaxJohnsen, 2019-05-15:
Are you required to use all the different vehicles? You may want to find the correct transformations for a subset of the blueprints, and only use these when spawning a hero actor.

morningsky, 2019-05-15:
@MaxJohnsen  Yes, you are right. I just use these when spawning a hero actor. Tesla model 3 is enough! Thanks! 
I have another problem maybe you can help me... How can I get the hero id when I run manual_control.py ?  I haven't found it in API...

closed 2019-05-22 by morningsky

Starchy0628, 2026-04-06:
跟着gemini搜过来的，想不到之前调试环境真的是在线求助，没有AI的日子是怎么过来的