carla-simulator/carla #1640: Applying VehicleControl(throttle=1.0) for the first time has a delay

DanielsJusts, 2019-05-14:
Hello!

What I noticed is that using `vehicle.apply_control(VehicleControl(throttle=1.0))` for the first time has quite a delay. Afterwards, the delay is not noticeable and the commands are immediate. This is causing me issues as I'm processing the sensor data and then sending back the necessary vehicle control signals. The delay could even be observed when running `manual_control.py` and trying to accelerate for the first time. As you can see, the vehicle is not reacting for a few seconds and then it runs as it should. The interesting part is that when I'm applying steering command alongside the throttle command (`vehicle.apply_control(VehicleControl(throttle=1.0, steer=1.0))`), it can be observed that the wheels turn immediately, but the throttle is applied with a delay.

What could be causing this delay?

Also, if there is no workaround for this delay, maybe there is some way that I can get feedback as to when the throttle is going to be applied as the commands are received and `vehicle.get_control()` returns that the throttle is applied, but in the server physically the vehicle is not yet moving. I know that an alternative could be to use synchronous mode, but I'm just wondering if there are other options.

If it is of any concern, I am using Carla 0.9.5 on Ubuntu 16.10.

oldtopos, 2019-05-17:
I am seeing this as well with 0.9.5 on Ubuntu 18.04.  I plotted output through the ROS bridge with rqt_plot and the delay is obvious.  It happens with every vehicle after spawning.

marcgpuig, 2019-05-20:
This delay is caused by Unreal car's system, where they somehow simulate the time that the driver will spend starting the car engine or changing a new gear. I think one solution is to spawn a car, change the gear system to manual, put the first gear, and then set the gear system to automatic again.

oldtopos, 2019-05-22:
Marc is this documented somewhere in UE4 ?  I am getting zero Google hits ...

oldtopos, 2019-05-24:
I created the simple vehicle project directly in UE4.  I sense no delay in vehicle movement using the UE4 player directly with the sample project

DanielsJusts, 2019-05-27:
Hello, @oldtopos and @marcgpuig and thank you both for your input!
@marcgpuig, I played around in Carla and your workaround seems to work.

Derekabc, 2019-07-03:
@DanielsJusts, can you explain how to avoid the delay in details?

DanielsJusts, 2019-07-04:
Hello @Derekabc!
I think @marcgpuig stated this pretty well in his comment:
> I think one solution is to spawn a car, change the gear system to manual, put the first gear, and then set the gear system to automatic again.

Basically, after spawning a car you need to switch from automatic transmission to manual transmission and put the car in the first gear (`vehicle.apply_control(VehicleControl(manual_gear_shift=True, gear=1))`) and then you switch back to automatic transmission (`vehicle.apply_control(VehicleControl(manual_gear_shift=False))`). Then, as much as tested, there is no delay when applying throttle for the first time.

linxigjs, 2019-07-12:
hi, if i use throttle to control vehicle's velocity, what's the function relationship between throttle and velocity?  Where is the code? And what's the difference between throttle and acceleration? Acctually, i want to use acceleration to control the velocity in carla. Thanks.

closed 2019-09-17 by stale[bot]

YenTran32, 2021-02-12:
In Windows, the method of switching between automatic and manual gear worked. However, the delay about 2 seconds (and depending on the amount of applied throttle) still existed in Ubuntu 18.04 and Carla 0.9.10 or 0.9.11. If manual gear (e.g., 1) is used, there's no delay. But the delay always appeared if using automatic gear (for the first time of any run). Does anyone have a solution for this?

sinamoradi1989, 2022-05-04:
Guys anyone knows how to relate throttle to acceleration in Carla? I just know it varies between [0 1].
I'm developing NMPC control and I have figured out the lateral dynamics. Yet, I don't know about the longitudinal and how the throttle adjustment affects the vehicle.
Once I saw someone saying it has 3 default settings i.e. 0,0.5,1 which are neutral, for maintaining speed, and full throttle, respectively. This is very weird to me.