carla-simulator/carla #1641: Map.get_waypoint/3

d0m3k, 2019-05-14:
Using CARLA 0.9.5 on Ubuntu 16.04, on map `Town05`:

I have an issue regarding usage of new Waypoint API with points of types other than `carla.LaneType.Driving`, specifically `carla.LaneType.Sidewalk`.

It seems that, while since 0.9.5 there is version of `Map.get_waypoint` with parameter `lane_type`, when `project_to_road` is set to `False` and type other than `Driving` is chosen, it always returns `None` instead of waypoint, effectively blocking from fetching paths for walkers.
(See `PythonAPI/examples/pedestrian_automatic.py` and related `PythonAPI/carla/agents/navigation/pedestrian_agent.py` on branch `pedestrian-walking` [in my fork](https://github.com/d0m3k/carla/tree/pedestrian-walking)).

Is this lack of implementation, or my misuse? I see that OpenDrive `xodr` files of built-in maps have information on sidewalks which is parsed by Carla, yet both `Map.get_topology` [1] and `Map.get_waypoint` methods seem to omit them completely.

[1] -- on provided `pedestrian-walking` branch there is a slightly modified version of `Map.cpp`, which includes sidewalks in topology -- there is still no possibility to fetch any sidewalk-based waypoint.

marcgpuig, 2019-05-20:
Hi @d0m3k 

`project_to_road=False` is used to determine if a location, where you asked for a waypoint, is inside the lane (road lane, sidewalk, shoulder...) of the nearest waypoint (of the type `lane_type`), if the location is outside the lane, the result of this waypoint is `None`. This is useful, for instance, if you want to know if your car is inside the road.
Could be, that you get `None` because these locations that you are asking, they are not inside the lane itself?

d0m3k, 2019-05-27:
My case was using pedestrians spawned on spawn points dedicated for cars, so, if it should return `None` when no `Sidewalk`-type waypoint is available in close proximity, what you describe is valid; the script asks for waypoint based on current pedestrian location (and that would fail, for it's walking in the middle of road dedicated for cars).

So if I understand correctly, `project_to_road` set to `True` should try to find the closest path available _on the lane of required type_? Because from what I saw during my tests, whatever value I set as `lane_type`, when `project_to_road=True`, waypoints were exactly the same as for `Driving` lane, at least on `0.9.5` on `Town05`. (I'll try it again to be sure I'm not doing anything stupid when project rebuilds)

d0m3k, 2019-05-28:
UPDATE #2: There must be an error in my implementation, because I just modified `lane_explorer.py` and successfully rendered data on sidewalks; there are no junctions between sidewalks for them, though. 
Lemme close it for time of rewrite; thank you for your help!

closed 2019-05-28 by d0m3k