carla-simulator/carla #1642: Carla 0.9.3 pedestrians control

fty0724, 2019-05-15:
Hello!
As I saw in the new version video in Carla 0.9.3, I can see the pedestrian is crossing the street when the car just arrived.
So I want to know how to realize this? That not only the pedestrian but also the other NPC cars can emerge at the right time I want.

dixantmittal, 2019-05-21:
Hi there!

As I understand the question, one possible way to spawn pedestrian when car arrives can be to connect multiple clients to the carla server. When the player car arrives, spawn NPC or pedestrian using another client (a separate thread/process). Would be easier if you could re-frame your question. What precisely do you want to do?

fty0724, 2019-05-22:
Hello!
You have got what I mean, and I thanks for your idea of using multiple clients, and I also want to know how to control the right time,how can the pedestrian client know when to start as the player car arrives?

dixantmittal, 2019-07-22:
Hi @fty0724

You might want to use a distance-based pedestrian starter. You can get the autonomous car's position by using ```world.get_actors()```. 
Then,
```
 if magnitude(npc.get_location() - av.get_location()) < threshold:
   npc.start_moving()
```

_Dixant_

fty0724, 2019-07-26:
Thanks for your reply! I will try it later~

closed 2019-10-01 by stale[bot]