carla-simulator/carla #1645: spawn_npc.py Crashes the Simulator

tianhao-david-hu, 2019-05-15:
The demo spawn_npc.py used to work, but in the latest commit cf58f7a2c, it will crash the simulator.
This is output on the terminal prior to the crash:
--------------------------------------------------------------------------------------------------------
[2019.05.15-13.08.35:729][ 89]LogCarla: Spawning actor 'vehicle.nissan.patrol'
[2019.05.15-13.08.35:734][ 89]LogCarla: Spawning actor 'vehicle.gazelle.omafiets'
[2019.05.15-13.08.35:737][ 89]LogCarla: Spawning actor 'vehicle.mini.cooperst'
[2019.05.15-13.08.35:738][ 89]LogBlueprintUserMessages: [Mini_C_0] MaterialNotFound
[2019.05.15-13.08.35:738][ 89]LogCarla: Spawning actor 'vehicle.dodge_charger.police'
[2019.05.15-13.08.35:739][ 89]LogCarla: Spawning actor 'vehicle.bmw.grandtourer'
[2019.05.15-13.08.35:740][ 89]LogCarla: Spawning actor 'vehicle.chevrolet.impala'
[2019.05.15-13.08.35:743][ 89]LogCarla: Spawning actor 'vehicle.lincoln.mkz2017'
[2019.05.15-13.08.35:744][ 89]LogCarla: Spawning actor 'vehicle.nissan.patrol'
[2019.05.15-13.08.35:745][ 89]LogCarla: Spawning actor 'vehicle.diamondback.century'
[2019.05.15-13.08.35:747][ 89]LogCarla: Spawning actor 'vehicle.nissan.patrol'
Failed to find symbol file, expected location:
"/home/david/carla/Dist/0.9.5-95-gcf58f7a/LinuxNoEditor/CarlaUE4/Binaries/Linux/CarlaUE4.sym"
Assertion failed: IsInGameThread() [File:/home/david/UnrealEngine_4.22/Engine/Source/Runtime/Engine/Classes/Animation/AnimInstance.h] [Line: 1241] 

Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=103376 LargeMemoryPoolOffset=234512 
[2019.05.15-13.08.35:772][ 89]LogCore: **Error: appError called: Assertion failed: IsInGameThread()** [File:/home/david/UnrealEngine_4.22/Engine/Source/Runtime/Engine/Classes/Animation/AnimInstance.h] [Line: 1241] 

[2019.05.15-13.08.35:778][ 89]LogCore: === Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to read memory at address 0x00000000000040b1

[2019.05.15-13.08.35:778][ 89]LogCore: Assertion failed: IsInGameThread() [File:/home/david/UnrealEngine_4.22/Engine/Source/Runtime/Engine/Classes/Animation/AnimInstance.h] [Line: 1241] 

0x0000000004e5aeef CarlaUE4!UnknownFunction(0x4c5aeee)
0x0000000004e2bca0 CarlaUE4!UnknownFunction(0x4c2bc9f)
0x0000000004b20bac CarlaUE4!UnknownFunction(0x4920bab)
0x0000000004cbc180 CarlaUE4!UnknownFunction(0x4abc17f)
0x000000000249f64b CarlaUE4!UnknownFunction(0x229f64a)
0x00000000024a5f5a CarlaUE4!UnknownFunction(0x22a5f59)
0x0000000005bcdfdf CarlaUE4!UnknownFunction(0x59cdfde)
0x00007ffb25a21830 libc.so.6!__libc_start_main(+0xef)
0x0000000002495029 CarlaUE4!UnknownFunction(0x2295028)

[2019.05.15-13.08.35:790][ 89]LogExit: Executing StaticShutdownAfterError
[2019.05.15-13.08.35:800][ 89]LogCore: Warning: Unable to statfs('/home/david/carla/Dist/0.9.5-95-gcf58f7a/LinuxNoEditor/CarlaUE4/Saved/Crashes/crashinfo-CarlaUE4-pid-16561-0BD4434808D6D914E966EB883C33B792/CarlaUE4.log'): errno=2 (No such file or directory)
Malloc Size=512385 LargeMemoryPoolOffset=746913 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)

nsubiron, 2019-05-16:
I'm not able to reproduce, have you tried with the last [nightly build](http://carla-assets-internal.s3.amazonaws.com/Releases/Linux/Dev/CARLA_Latest.tar.gz)?

Xiaoyu006, 2019-05-17:
> ## The demo spawn_npc.py used to work, but in the latest commit [cf58f7a](https://github.com/carla-simulator/carla/commit/cf58f7a2c1a57b344a0a033b8522823206c901c0), it will crash the simulator.
> This is output on the terminal prior to the crash:
> [2019.05.15-13.08.35:729][ 89]LogCarla: Spawning actor 'vehicle.nissan.patrol'
> [2019.05.15-13.08.35:734][ 89]LogCarla: Spawning actor 'vehicle.gazelle.omafiets'
> [2019.05.15-13.08.35:737][ 89]LogCarla: Spawning actor 'vehicle.mini.cooperst'
> [2019.05.15-13.08.35:738][ 89]LogBlueprintUserMessages: [Mini_C_0] MaterialNotFound
> [2019.05.15-13.08.35:738][ 89]LogCarla: Spawning actor 'vehicle.dodge_charger.police'
> [2019.05.15-13.08.35:739][ 89]LogCarla: Spawning actor 'vehicle.bmw.grandtourer'
> [2019.05.15-13.08.35:740][ 89]LogCarla: Spawning actor 'vehicle.chevrolet.impala'
> [2019.05.15-13.08.35:743][ 89]LogCarla: Spawning actor 'vehicle.lincoln.mkz2017'
> [2019.05.15-13.08.35:744][ 89]LogCarla: Spawning actor 'vehicle.nissan.patrol'
> [2019.05.15-13.08.35:745][ 89]LogCarla: Spawning actor 'vehicle.diamondback.century'
> [2019.05.15-13.08.35:747][ 89]LogCarla: Spawning actor 'vehicle.nissan.patrol'
> Failed to find symbol file, expected location:
> "/home/david/carla/Dist/0.9.5-95-gcf58f7a/LinuxNoEditor/CarlaUE4/Binaries/Linux/CarlaUE4.sym"
> Assertion failed: IsInGameThread() [File:/home/david/UnrealEngine_4.22/Engine/Source/Runtime/Engine/Classes/Animation/AnimInstance.h] [Line: 1241]
> 
> Signal 11 caught.
> Malloc Size=65538 LargeMemoryPoolOffset=65554
> CommonUnixCrashHandler: Signal=11
> Malloc Size=65535 LargeMemoryPoolOffset=131119
> Malloc Size=103376 LargeMemoryPoolOffset=234512
> [2019.05.15-13.08.35:772][ 89]LogCore: **Error: appError called: Assertion failed: IsInGameThread()** [File:/home/david/UnrealEngine_4.22/Engine/Source/Runtime/Engine/Classes/Animation/AnimInstance.h] [Line: 1241]
> 
> [2019.05.15-13.08.35:778][ 89]LogCore: === Critical error: ===
> Unhandled Exception: SIGSEGV: invalid attempt to read memory at address 0x00000000000040b1
> 
> [2019.05.15-13.08.35:778][ 89]LogCore: Assertion failed: IsInGameThread() [File:/home/david/UnrealEngine_4.22/Engine/Source/Runtime/Engine/Classes/Animation/AnimInstance.h] [Line: 1241]
> 
> 0x0000000004e5aeef CarlaUE4!UnknownFunction(0x4c5aeee)
> 0x0000000004e2bca0 CarlaUE4!UnknownFunction(0x4c2bc9f)
> 0x0000000004b20bac CarlaUE4!UnknownFunction(0x4920bab)
> 0x0000000004cbc180 CarlaUE4!UnknownFunction(0x4abc17f)
> 0x000000000249f64b CarlaUE4!UnknownFunction(0x229f64a)
> 0x00000000024a5f5a CarlaUE4!UnknownFunction(0x22a5f59)
> 0x0000000005bcdfdf CarlaUE4!UnknownFunction(0x59cdfde)
> 0x00007ffb25a21830 libc.so.6!__libc_start_main(+0xef)
> 0x0000000002495029 CarlaUE4!UnknownFunction(0x2295028)
> 
> [2019.05.15-13.08.35:790][ 89]LogExit: Executing StaticShutdownAfterError
> [2019.05.15-13.08.35:800][ 89]LogCore: Warning: Unable to statfs('/home/david/carla/Dist/0.9.5-95-gcf58f7a/LinuxNoEditor/CarlaUE4/Saved/Crashes/crashinfo-CarlaUE4-pid-16561-0BD4434808D6D914E966EB883C33B792/CarlaUE4.log'): errno=2 (No such file or directory)
> Malloc Size=512385 LargeMemoryPoolOffset=746913
> Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
> Segmentation fault (core dumped)

I got the same problem when I was using nvidia driver 384 with 2 GTX 1080 GPUs. And this is fixed after updating the nvidia driver from 384 to 410. 
I hopt it helps!

By the way, when the computer is updating nvidia driver, I went to dinner with my friend. What unbelievable is that, we found that all the files of UE4 and carla are deleted from my computer. I still do not find out why it just happens.

tianhao-david-hu, 2019-05-24:
I ended up with downgrading to carla 0.9.5 version from the nightly build. The problem was solved. 
If I notice anything new, I'll post it here.

closed 2020-06-26 by germanros1987