carla-simulator/carla #1646: Town02 with invalid RoadOption

rohjunha, 2019-05-15:
[image]

I am generating trajectories with RoamingAgent in the example code with Carla 0.9.5. I found that this specific intersection in Town02 gives me wrong RoadOption.LEFT instead of RoadOption.STRAIGHT or RoadOption.RIGHT. Is it a bug in the map?

Derekabc, 2019-06-02:
Any update to this problem?

rohjunha, 2019-06-20:
@Derekabc I think this is a bug of the code deciding the direction.
This is the part that the local planner decides the connection type: https://github.com/carla-simulator/carla/blob/master/PythonAPI/carla/agents/navigation/local_planner.py#L280-L292
I fixed the code like this:
```
n = next_waypoint.transform.rotation.yaw
n = n % 360.0

c = current_waypoint.transform.rotation.yaw
c = c % 360.0

n -= 180.0
c -= 180.0
assert -180 <= n < 180
assert -180 <= c < 180
diff_angle = n - c
diff_angle += -360 if diff_angle > 180 else (360 if diff_angle < -180 else 0)

angle_threshold = 5

str_new = None
if abs(diff_angle) < angle_threshold:
    return RoadOption.STRAIGHT
elif diff_angle < 0.0:
    return RoadOption.LEFT
else:
    return RoadOption.RIGHT
```

closed 2019-08-26 by stale[bot]