carla-simulator/carla #1648: Retrieve topological planner high-level commands for CARLA 0.9.5

arunk786, 2019-05-16:
I am trying to get the high level commands from topological or global planner in CARLA 0.9.5.
CARLA 0.9.5 release documentation ([http://carla.org/2019/04/03/release-0.9.5/](http://carla.org/2019/04/03/release-0.9.5/) shows GlobalRoutePlannerDAO to get high-level actions.
1. I am unable to find GlobalRoutePlannerDAO
2. I am trying to train a new model to learn driving(throttle, steer and break). (a) So, i want to access high-level command (TURN LEFT, TURN RIGHT) at every step before applying control for a learning "agent". Is there any way to obtain it ?
3. Is "LANE CHANGE LEFT/RIGHT" a part of high-level command for an agent, if any ?

closed 2019-07-22 by stale[bot]

weizhaoji, 2021-03-14:
Hi，I have the same problem. How did you solve this problem?

MalekWahidi, 2024-01-01:
Same problem here. Couldn't find documentation on how to access high-level commands for stuff like conditional imitation learning where the AI agent is conditioned by a one-hot vector (for example: [Straight, Right, Left, etc.]), in addition to sensor inputs, to figure out what to do at intersections and how to reach desired destinations. I need to collect training data from the CARLA autopilot but I am not sure if and how one could get these high-level commands that are guiding the autopilot on a frame-by-frame basis.