carla-simulator/carla #1651: [Buid carla on Ubuntu]make launch error

jd401, 2019-05-18:
Hello,
I am trying to build Carla on Ubuntu 16.04.
i followed all steps from the official document https://carla.readthedocs.io/en/latest/how_to_build_on_linux/.However i get stuck at “make 
launch” step,following errors like:
------------------------------------------------------------------------------------------------------------------------------
a401@401:~/carla$ sudo make launch
Setup.sh: llvm-7.0 already installed.
Setup.sh: boost-1.69.0-c7 already installed.
Setup.sh: rpclib-v2.2.1_c1-c7 already installed.
Setup.sh: gtest-1.8.1-c7 already installed.
Setup.sh: CARLA version 0.9.5-96-g67cfd574-dirty.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Server.Release" configuration.
ninja: no work to do.
[0/1] Install the project...
-- Install configuration: "Server"
BuildLibCarla.sh: Success!
BuildCarlaUE4.sh: Using Unreal Engine at '/home/a401/UnrealEngine_4.22'
BuildCarlaUE4.sh: Build CarlaUE4 project.
make[1]: Entering directory '/home/a401/carla/Unreal/CarlaUE4'
bash "/home/a401/UnrealEngine_4.22/Engine/Build/BatchFiles/Linux/RunMono.sh" "/home/a401/UnrealEngine_4.22/Engine/Binaries/DotNET/UnrealBuildTool.exe" CarlaUE4Editor Linux Development  -project="/home/a401/carla/Unreal/CarlaUE4/CarlaUE4.uproject" 

Running Mono...

Fixing inconsistent case in filenames.
Setting up Mono
~/UnrealEngine_4.22/Engine ~/carla/Unreal/CarlaUE4
Using 'git status' to determine working set for adaptive non-unity build (/home/a401/UnrealEngine_4.22).
Creating makefile for CarlaUE4Editor (no existing makefile)
Parsing headers for CarlaUE4Editor
  Running UnrealHeaderTool "/home/a401/carla/Unreal/CarlaUE4/CarlaUE4.uproject" "/home/a401/carla/Unreal/CarlaUE4/Intermediate/Build/Linux/B4D820EA/CarlaUE4Editor/Development/CarlaUE4Editor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors
**Refusing to run with the root privileges.**
Makefile:215: recipe for target 'CarlaUE4Editor' failed
make[1]: *** [CarlaUE4Editor] Error 1
make[1]: Leaving directory '/home/a401/carla/Unreal/CarlaUE4'
Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
make: *** [launch] Error 2
----------------------------------------------------------------------------------------------------------------------------
i can't figure out the error of "Refusing to run with the root privileges.",i can't do "make launch" without root privilege,anyone knowing how to solve this issue please kindly help me. :(

nsubiron, 2019-05-20:
Try doing a new clone of the project without sudo.

closed 2019-07-26 by stale[bot]

SyedUmerHasan, 2019-10-05:
heyyy!
Did you find any solution to this?
I have same problem

roserbatlle, 2019-12-19:
Hi @jd401 @SyedUmerHasan! 
Did you find a solution regarding the problem discussed above? 
I'm currently facing a similar problem! 
Thanks!

roser-br, 2019-12-19:
Hi @jd401 @SyedUmerHasan! 
Did you find a solution regarding the problem discussed above? 
I'm currently facing a similar problem! 
Thanks!