carla-simulator/carla #1652: Can I change spawned vehicles speed?

SangDooLee, 2019-05-19:
It is not a issue but I have a something curious.

I just want to do Reinforcement Learning with Carla. I would like to train my car with reckless driving cars in the road.

Can I change spawned vehicles speed?

dixantmittal, 2019-05-21:
Hi there!

Yes, you can change the speed of any car by calling actor's apply_control() function. Just pass the desired throttle values in the VehicleControl object.

```
actor = world.try_spawn_actor(bp, transform)
actor.apply_control(carla.VehicleControl(throttle = np.random.rand(), brake = np.random.rand()))
```

This small code should generate random driving behaviour for non-player actors. 

- Dixant

kraken24, 2019-07-03:
Hey everyone, I have initialized a vehicle with speed of 10 m/s. Now if I apply carla.vehicleControl(throttle=0, brake=0) to get constant velocity, it fails and vehicle eventually stops. How can I move vehicle at constant speed?

Derekabc, 2019-07-03:
this command give no throttle to the vehicle. The value of Throttle is between [0, 1].  set_velocity(vehicle.get_velocity()) may be also helpful for you to get a constant speed.

dixantmittal, 2019-07-04:
Hi @kraken24,

If you don't apply any throttle, the car will ultimately stop due to engine brakes, as it mimics actual vehicle dynamics. You can see the carla.VehiclePhysicsControl to know the detailed physics of the car you are using.

- Dixant

linxigjs, 2019-07-12:
What should i do if i want to control the velocity by  adjusting the accelaration(throttle)? What's the function relationship between throttle and velocity? @Derekabc @dixantmittal Thanks.

kimiya66, 2019-07-19:
@ Dixant Is throttle the only parameter to control vehicle's speed ? I mean can I just diectly change velocity by setting set_velocity(). However, I used this function set_velocity and vehicle's speed does not change at all! after I set it to some magnitude when I call vehicle.get_velocity() , I see that nothing changed and the value is still zero!!
Thank you for your help in advance if anyone know hoe to control velocity I would appreciate it.

dixantmittal, 2019-07-20:
Hi @linxigjs,
The relation between velocity and throttle is highly non-linear and difficult to map. What you can do is implement a small PID controller to get the desired velocity by nudging the throttle value to x + Δ if velocity < v else x - Δ. This is the easiest way I found to implement it. You can also use set_velocity() but the velocity of the car decreases over time due to engine brakes. 

_Dixant_

dixantmittal, 2019-07-20:
Hi @kimiya66,
set_velocity() is working fine for me. Although one caution, the operations set_velocity() and get_velocity() are not instantaneous. There is a small lag. Maybe you would want to add a little delay before calling get_velocity(). 

_Dixant_

kimiya66, 2019-07-20:
> Hi @kimiya66,
> set_velocity() is working fine for me. Although one caution, the operations set_velocity() and get_velocity() are not instantaneous. There is a small lag. Maybe you would want to add a little delay before calling get_velocity().
> 
> _Dixant_

yes it can be because of the delay. however I added function vehicle.get_physics_control() before calling set_velocity and I saw some changes in velocity(through vehicle.get_velocity()). it shows that vehicle get some values for velocity but they were not the exact values that I set with vehicle.set_velocity()!!
I just want to make sure that what is the exact parameter to see vehicle's current speed(is it just get_velocity() or are there other ways?) 
Thank you

dixantmittal, 2019-07-22:
set_velocity(x) works by setting the velocity of the car to x only for an instant. If you're not applying throttle and the vehicle is in gear, its engine brakes will decrease its velocity (just like in a real car). 
If you want to maintain a constant velocity, you can do it by calling set_velocity(x) in a loop. E.g.

```
car.apply_control(carla.VehicleControl(throttle=0, brake=0, manual_gear_shift=True, gear=0))
while True:
    car.set_velocity(carla.Vector3D(x,y,z))
    time.sleep(0.01)
    print(car.get_velocity().x, car.get_velocity().y)
```

You should be able to get an almost constant velocity (x ± ϵ)

_Dixant_

closed 2019-09-27 by stale[bot]

SLR-96, 2022-01-07:
The problem is that although you can set the vehicle's velocity, you cannot set its longitudinal speed (the speed you would get on the speedometer). And since in RL you don't know the random orientations you will get at each time step, it's very difficult to set the correct velocity vector for the vehicle's random speed.

niezifei, 2023-07-21:
e.g,... In generate_traffic.py, you could change the `vehicle_percentage_speed_difference(self, actor, percentage)`, to set the difference between expected vehicle speed and current road speed limitation. By setting the percentage as a negative value, the vehicle speed could violate the road speed limits.