carla-simulator/carla #1654: Fatal Error while executing generate_map.py for importing new maps in carla

Dhairyashil-kpit, 2019-05-20:
Dear all,

I am new to Carla and simulation software. I am trying to import new maps in carla_9.5, I refer many issues related to import maps in carla, but I didn't got any satisfactory process for importing new maps in carla.

There are two process to import maps in carla:
1. Manually load maps via unreal engine ref: https://carla.readthedocs.io/en/latest/how_to_make_a_new_map/
2. Automatically load maps via python script: ref: https://carla.readthedocs.io/en/latest/generate_map_from_fbx/

I followed Automatic process load maps in carla using **generate_map.py** script:

1. Downloaded sample maps from OpenDrive : http://www.opendrive.org/tools/Crossing8Course.zip
2. Generated the FilmBox .fbx files using above xodr file using RoadRunner.
3. Then created directory with same name as xodr and fbx files inside RoadRunnerFiles of carla repo.
4. Then executed the python script: **Util/generate_map.py**
5. and getting **_Fatal error!!_** please find attached log files with screen-shot

```
Building Assets For Linux
Guid format shader working directory is 15 characters bigger than the processId version (../../../../Documents/carla_9_5/carla/Unreal/CarlaUE4/Intermediate/Shaders/WorkingDirectory/6079/).
Cleaned the shader compiler working directory '../../../../Documents/carla_9_5/carla/Unreal/CarlaUE4/Intermediate/Shaders/tmp/000426F008D6DD2D3D50E8E09444FB9D/'.
Using Local Shader Compiler.
Max Cache Size: 512 MB
Loaded boot cache 0.09s 155MB ../../../../Documents/carla_9_5/carla/Unreal/CarlaUE4/DerivedDataCache/Boot.ddc.
Loaded Boot cache: ../../../../Documents/carla_9_5/carla/Unreal/CarlaUE4/DerivedDataCache/Boot.ddc
FDerivedDataBackendGraph:  Pak pak cache file ../../../../Documents/carla_9_5/carla/Unreal/CarlaUE4/DerivedDataCache/DDC.ddp not found, will not use a pak cache.
Unable to find inner node Pak for hierarchical cache Hierarchy.
FDerivedDataBackendGraph:  EnginePak pak cache file ../../../Engine/DerivedDataCache/DDC.ddp not found, will not use a pak cache.
Unable to find inner node EnginePak for hierarchical cache Hierarchy.
Using Local data cache path ../../../Engine/DerivedDataCache: Writable
Shared data cache path not found in *engine.ini, will not use an Shared cache.
Unable to find inner node Shared for hierarchical cache Hierarchy.
Verifying Global Shaders for GLSL_430
FAssetRegistry took 0.0021 seconds to start up
Selected Device Profile: [Linux]
Using QuadricMeshReduction for automatic static mesh reduction
Using SimplygonMeshReduction for automatic skeletal mesh reduction
No automatic mesh merging module available
No distributed automatic mesh merging module available
No automatic mesh merging module available
No distributed automatic mesh merging module available
CarlaUE4 1.0.0, NetCL: 0, EngineNetVer: 10, GameNetVer: 0 (Checksum: 1844717674)
Processed 9 localized package path(s) for 1 prioritized culture(s) in 0.016016 seconds
Linux SourceCodeAccessSettings: NullSourceCodeAccessor
Loaded 0 collections in 0.001694 seconds
Scanning file cache for directory '/home/kpit/Documents/carla_9_5/carla/Unreal/CarlaUE4/Saved/Collections/' took 0.00s
Scanning file cache for directory '/home/kpit/Documents/carla_9_5/carla/Unreal/CarlaUE4/Content/Developers/kpit/Collections/' took 0.00s
Scanning file cache for directory '/home/kpit/Documents/carla_9_5/carla/Unreal/CarlaUE4/Content/Collections/' took 0.00s
Rebuilt the object cache for 0 collections in 0.000000 seconds (found 0 objects)
Fixed up redirectors for 0 collections in 0.000042 seconds (updated 0 objects)
Can't find file '/Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapRoad'
Can't find file '/Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapRoad'
Can't find file '/Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapSideWalkCurb'
Can't find file '/Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapSideWalkCurb'
Can't find file '/Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapSideWalk_00'
Can't find file '/Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapSideWalk_00'
Can't find file '/Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapLaneMarking'
Can't find file '/Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapLaneMarking'
Can't find file '/Game/Carla/Static/GenericMaterials/WetPavement/WetPavement_Complex_Road_N2'
Can't find file '/Game/Carla/Static/GenericMaterials/WetPavement/WetPavement_Complex_Road_N2'
Can't find file '/Game/Carla/Static/GenericMaterials/WetPavement/WetPavement_Complex_Concrete'
Can't find file '/Game/Carla/Static/GenericMaterials/WetPavement/WetPavement_Complex_Concrete'
Can't find file '/Game/Carla/Static/GenericMaterials/Ground/SideWalks/SidewalkN4/WetPavement_SidewalkN4'
Can't find file '/Game/Carla/Static/GenericMaterials/Ground/SideWalks/SidewalkN4/WetPavement_SidewalkN4'
Can't find file '/Game/Carla/Static/GenericMaterials/LaneMarking/Lanemarking'
Can't find file '/Game/Carla/Static/GenericMaterials/LaneMarking/Lanemarking'
== CARLA Settings ==============================================================
Last settings file loaded: 
[CARLA/Server]
RPC Port = 2000
Streaming Port = 2001
Server Time-out = 10000 ms
Synchronous Mode = Disabled
Rendering = Enabled
[CARLA/QualitySettings]
Quality Level = Epic
================================================================================
Source control is disabled
Source control is disabled
Source control is disabled
Source control is disabled
47501 objects as part of root set at end of initial load.
8672800 out of 0 bytes used by permanent object pool.
CloseDisregardForGC: 0/0 objects in disregard for GC pool
Executing Class /Script/Carla.MapProcessCommandlet
Initializing Game Engine...
Initializing Engine...
UGameplayTagsManager::InitializeManager -  0.000 s
Initializing FReadOnlyCVARCache
Set ProjectVersion to 0.9.5. Version Checksum will be recalculated on next use.
Saved boot cache 0.08s 155MB ../../../../Documents/carla_9_5/carla/Unreal/CarlaUE4/DerivedDataCache/Boot.ddc.
Texture streaming: Disabled
== CARLA Settings ==============================================================
Last settings file loaded: 
[CARLA/Server]
RPC Port = 2000
Streaming Port = 2001
Server Time-out = 10000 ms
Synchronous Mode = Disabled
Rendering = Enabled
[CARLA/QualitySettings]
Quality Level = Epic
================================================================================
Game Engine Initialized.
Initializing Game Engine Completed
=== Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to read memory at address 0x00000000000017c8

Fatal error!

0x00007fcea81fbc6a libUE4Editor-Engine.so!FSubsystemCollectionBase::GetSubsystemInternal(TSubclassOf<USubsystem>) const [/home/kpit/UnrealEngine_4.22/Engine/Source/Runtime/Engine/Private/Subsystems/SubsystemCollection.cpp:52]
0x00007fce56c3fa9a libUE4Editor-Paper2DEditor.so!FPaper2DEditor::OnPostEngineInit() [/home/kpit/UnrealEngine_4.22/Engine/Plugins/2D/Paper2D/Source/Paper2DEditor/Private/Paper2DEditorModule.cpp:184]
0x00007fce56c40cbf libUE4Editor-Paper2DEditor.so!TBaseRawMethodDelegateInstance<false, FPaper2DEditor, void ()>::ExecuteIfSafe() const [/home/kpit/UnrealEngine_4.22/Engine/Source/Runtime/Core/Public/Delegates/DelegateInstancesImpl.h:516]
0x0000000000255cf2 UE4Editor!TBaseMulticastDelegate<void>::Broadcast() const [/home/kpit/UnrealEngine_4.22/Engine/Source/Runtime/Core/Public/Delegates/DelegateSignatureImpl.inl:977]
0x00000000002364e5 UE4Editor!FEngineLoop::PreInit(char16_t const*) [/home/kpit/UnrealEngine_4.22/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:2682]
0x0000000000249ab2 UE4Editor!GuardedMain(char16_t const*) [/home/kpit/UnrealEngine_4.22/Engine/Source/Runtime/Launch/Private/Launch.cpp:129]
0x00007fceab54b149 libUE4Editor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*)) [/home/kpit/UnrealEngine_4.22/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:243]
0x00007fcea0296830 libc.so.6!__libc_start_main(+0xef)
0x0000000000231029 UE4Editor!_start()

Executing StaticShutdownAfterError
Unable to statfs('/home/kpit/Documents/carla_9_5/carla/Unreal/CarlaUE4/Saved/Crashes/crashinfo-CarlaUE4-pid-6079-04CCD6F808D6DD2D44E44C149506EBC9/CarlaUE4.log'): errno=2 (No such file or directory)
```

I am not able to understand the proper process for importing new maps in carla. So please help me. 

Thanks in advanced.

dnovillo-CVC, 2019-05-20:
Hi @Dhairyashil-kpit ! I'll look into it. Could you provide with the fbx so we can reproduce the exact case?

Dhairyashil-kpit, 2019-05-21:
Hi @dnovillo-CVC ,

I work with Dhairyashil. Kindly refer the attached .tar.gz file which includes :
1. .fbx file
2. .xodr file

However, I am not able to attach the assets as it is exceeding the size limit. 
Kindly use the .xodr file and open it in RoadRunner and the export it into .fbx to get all assets.

[sample_export_from_roadrunner.tar.gz](https://github.com/carla-simulator/carla/files/3200777/sample_export_from_roadrunner.tar.gz)

dnovillo-CVC, 2019-05-22:
Hi @Dhairyashil-kpit . Thanks for your feedback. We have, potentially, solved the issue. We are submitting a PR asap to add the code fixes.

yamrodest, 2019-06-17:
Hi @dnovillo-CVC, Could you provide more or less the date of the PR? I have the same problem and I would like to know when approx. it will be fixed. Thanks!

closed 2020-06-26 by germanros1987