carla-simulator/carla #1655: IndexError

vishal-prgmr, 2019-05-20:
Carla 0.9.5
Windows 7
Python 3.7.3

Built carla from scratch, packaged the project with CarlaGameModeBase. Ran 
**CarlaUE4.exe -carla-port=2030**
[image]

The set ports are opened.

[image]

Now running manual_control.py gives the following error which means the blueprints are not fetched from the server. 

How do i solve this?

[image]

````

(caarla) C:\carla\PythonAPI\examples>python spawn_npc.py -n 80

destroying 0 actors

done.
Traceback (most recent call last):
  File "spawn_npc.py", line 129, in <module>
    main()
  File "spawn_npc.py", line 102, in main
    blueprint = random.choice(blueprints)
  File "C:\Users\TEXVNQA\AppData\Local\Continuum\anaconda3\envs\caarla\lib\rando
m.py", line 261, in choice
    raise IndexError('Cannot choose from an empty sequence') from None
IndexError: Cannot choose from an empty sequence

nsubiron, 2019-05-20:
Hi @vishal-prgmr,

You should use the "CarlaGameMode" blueprint instead of the "CarlaGameModeBase", and make sure there are actor factories registered
[image]

vishal-prgmr, 2019-05-20:
@nsubiron , How do i register the actor factories?

nsubiron, 2019-05-20:
Go to edit the "CarlaGameMode" blueprint, under details you'll find that menu, simply adding items to that list works. But by default you should have those 4 factories already registered.

vishal-prgmr, 2019-05-20:
Packaging the project with this Mode just gives a blank screen after packaging.

nsubiron, 2019-05-20:
No idea, have you modified something else, like the default pawn?

vishal-prgmr, 2019-05-20:
Not that I know. But i will figure it out by may be rolling back the build. 
However thank you for those crucial changes to the game mode.

closed 2019-05-20 by vishal-prgmr