carla-simulator/carla #1658: .Xodr

kslee12, 2019-05-20:
Hello, everyone. I have a question.

Can I import my own .Xodr file without road runner?

athreyasridharan, 2019-05-20:
From what I understand, the .xodr is used for road information, but since carla runs on Unreal, you're gonna need to generate Unreal assets so carla can render them.
What you can do is request RoadRunner for a license. Another route might be to get a filmbox (.fbx) file and follow carla's guide on generating a map.
In my case, I was able to obtain an OpenSceneGraph file and convert it to fbx. It's still a WIP though.

kslee12, 2019-05-22:
athreyasridharan, Thank you for your kind answer.

I will try.

AravindChandradoss, 2019-05-31:
Hi @kslee12 can you share that open-source software (for generating .xodr).
btw, is it compatible with OpenStreetMap?

Thanks

kslee12, 2019-06-03:
@athreyasridharan, I am sorry for my delayed response.

OpenDrive is not a commercial format, but software to generate .xodr seems to be commercial in many cases. This link may be the best open source I've found :  https://gitlab.com/OpenRoadEd/OpenRoadEd.

Thanks.

xiaoyang666, 2019-07-10:
Hi@kslee12 
I would like to ask how the fbx file is generated. I want to start from the openstreetmap direction. The original file is an osm file. Thank you.

closed 2019-09-15 by stale[bot]

zc804351220, 2020-02-28:
> I want to start from the openstreetmap direction. The original file is an osm file. Thank you.
@kslee12 
Hello ,I am newcomer of carlar, i am stumble in this problem too,have you solved this problem?
my thought is:
Blender can conver .osm to.fbx,will it make sense?

ish-101, 2021-08-01:
> OpenSceneGraph

Hey @athreyasridharan. Can you send me a link to the CARLA guide on generating a map that you mentioned? Also, when you said that you were able to get the OpenSceneGraph file, did you mean you were able to generate an OpenSceneGraph file corresponding to a .xodr file given as input. What tool did you use for that? I am trying to programmatically generate a very simple map that contains just mesh & textures for the road network in a given .xodr file.

athreyasridharan, 2021-08-01:
> > OpenSceneGraph
> 
> Hey @athreyasridharan. Can you send me a link to the CARLA guide on generating a map that you mentioned?
I believe I was referring to this one:
https://carla.readthedocs.io/en/0.9.7/how_to_make_a_new_map/

> Also, when you said that you were able to get the OpenSceneGraph file, did you mean you were able to generate an OpenSceneGraph file corresponding to a .xodr file given as input
IIRC, the company I worked with at the time was able to use their mapping software to export an OSG file. 

I then used `osgconv` to convert that file to an fbx, which I could then import into Unreal.

> I am trying to programmatically generate a very simple map that contains just mesh & textures for the road network in a given .xodr file.
The best (but expensive) answer was RoadRunner, but I haven't kept up with Carla in nearly 2y. From a quick search, it looks like Carla now [supports OpenDrive maps natively without needing a mesh](https://carla.readthedocs.io/en/latest/adv_opendrive/)?