carla-simulator/carla #1662: Carla : 0.9.5 Importing my map in CARLA

Dhairyashil-kpit, 2019-05-21:
Hi,

I am trying to import my map in the Carla. Kindly refer my another issue 1654, where I got issues in Automatic method of importing maps.

Currently I am trying manual method as per : https://carla.readthedocs.io/en/latest/how_to_make_a_new_map/ 

In Step 3 - Loading OpenDrive into CARLA, I copied my .xodr file in Content/Carla/Maps/OpenDrive folder and dragged the OpenDriveActor inside the scene and hit the play button in Unreal Engine : 
1. It made my viewport of Unreal Engine blank
2. when I clicked Stop button I got following errors : 

Blueprint Runtime Error: "Accessed None trying to read property StoredDirectionalLightActor". Blueprint:  BP_Weather Function:  Light Intensity Graph:  LightIntensity Node:  Set Intensity
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Light Intensity Graph:  LightIntensity Node:  Set Intensity
Blueprint Runtime Error: "Accessed None trying to read property StoredSkyLight". Blueprint:  BP_Weather Function:  Light Intensity Graph:  LightIntensity Node:  Set Intensity
Blueprint Runtime Error: "Accessed None trying to read property StoredDirectionalLightActor". Blueprint:  BP_Weather Function:  Orient Light Graph:  OrientLight Node:  SetActorRotation
Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Colors determined by sun position
Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Scalar Parameter Value
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Scalar Parameter Value
Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Scalar Parameter Value
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Scalar Parameter Value
Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Scalar Parameter Value
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Scalar Parameter Value
Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Scalar Parameter Value
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Scalar Parameter Value
Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Cloud opacity
Blueprint Runtime Error: "Attempted to assign to None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Cloud opacity
Blueprint Runtime Error: "Accessed None trying to read property StoredSkyLight". Blueprint:  BP_Weather Function:  Light Color Graph:  LightColor Node:  Set Light Color
Blueprint Runtime Error: "Accessed None trying to read property StoredDirectionalLightActor". Blueprint:  BP_Weather Function:  Light Color Graph:  LightColor Node:  Set Light Color
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Light Color Graph:  LightColor Node:  Set Light Color
Blueprint Runtime Error: "Accessed None trying to read property StoredDirectionalLightActor". Blueprint:  BP_Weather Function:  Light Color Graph:  LightColor Node:  Set Enable Light Shaft Bloom
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Light Color Graph:  LightColor Node:  Set Enable Light Shaft Bloom
Blueprint Runtime Error: "Accessed None trying to read property StoredDirectionalLightActor". Blueprint:  BP_Weather Function:  Light Color Graph:  LightColor Node:  Set Bloom Scale
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Light Color Graph:  LightColor Node:  Set Bloom Scale
Blueprint Runtime Error: "Accessed None trying to read property StoredAtmosfericFog". Blueprint:  BP_Weather Function:  Update Fog Graph:  UpdateFog Node:  Set Start Distance
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Fog Graph:  UpdateFog Node:  Set Start Distance
Blueprint Runtime Error: "Accessed None trying to read property StoredAtmosfericFog". Blueprint:  BP_Weather Function:  Update Fog Graph:  UpdateFog Node:  Set Density Multiplier
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Fog Graph:  UpdateFog Node:  Set Density Multiplier
Blueprint Runtime Error: "Accessed None trying to read property StoredAtmosfericFog". Blueprint:  BP_Weather Function:  Update Fog Graph:  UpdateFog Node:  Set Fog Multiplier
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Fog Graph:  UpdateFog Node:  Set Fog Multiplier
Blueprint Runtime Error: "Accessed None trying to read property StoredAtmosfericFog". Blueprint:  BP_Weather Function:  Update Fog Graph:  UpdateFog Node:  Set Sun Multiplier
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Fog Graph:  UpdateFog Node:  Set Sun Multiplier
Blueprint Runtime Error: "Accessed None trying to read property StoredDirectionalLightActor". Blueprint:  BP_Weather Function:  Light Intensity Graph:  LightIntensity Node:  Set Intensity
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Light Intensity Graph:  LightIntensity Node:  Set Intensity
Blueprint Runtime Error: "Accessed None trying to read property StoredSkyLight". Blueprint:  BP_Weather Function:  Light Intensity Graph:  LightIntensity Node:  Set Intensity
Blueprint Runtime Error: "Accessed None trying to read property StoredDirectionalLightActor". Blueprint:  BP_Weather Function:  Orient Light Graph:  OrientLight Node:  SetActorRotation
Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Colors determined by sun position
Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Scalar Parameter Value
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Scalar Parameter Value
[... cut ...]
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Fog Graph:  UpdateFog Node:  Set Fog Multiplier
Blueprint Runtime Error: "Accessed None trying to read property StoredAtmosfericFog". Blueprint:  BP_Weather Function:  Update Fog Graph:  UpdateFog Node:  Set Sun Multiplier
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Fog Graph:  UpdateFog Node:  Set Sun Multiplier
Play in editor start time for /Game/Carla/Maps/UEDPIE_0_Town_crossroad 0.618

Please help me to resolve this issue.

Dhairyashil-kpit, 2019-05-24:
Hi,

I Finally Got it !! 

>  How To Import Map in CARLA :

1. Download a sample map (.xodr : OpenDRIVE format) from : http://www.opendrive.org/download.html

2. Open Road Runner application and import the downloaded map

3. Export the file as Unreal (.fbx + .xml)

4. Again export the file as .xodr (OpenDRIVE)

5. Copy the exported .xodr file at /home/yourname/carla/Unreal/CarlaUE4/Content/Carla/Maps/OpenDrive.

6. Import the Roadrunner Plugin into the CarlaUE4 Project :
7. 	You will find the plugin folder (on Ubuntu) in /usr/bin/VectorZero/Tools/Unreal/Plugins.
8. 	Copy the Roadrunner Plugin into the Folder /home/yourname/carla/Unreal/CarlaUE4/Plugins.
9. 	Build carla : https://carla.readthedocs.io/en/latest/how_to_build_on_linux/ and do make launch
10. 	Open the Plugin Settings Edit->Plugings
11. 	Click Enable Roadrunner and restart.

12. Add new level at : /home/yourname/carla/Unreal/CarlaUE4/Content/Carla/Maps/ with the same name as .xodr file's name

13. Open that level in viewport

14. Create a folder under : /home/yourname/carla/Unreal/CarlaUE4/Content/ with the map name

15. Import .fbx in that folder

16. It will generate map in view port

17. Search for OpenDrive Actor and add it on the map

18. Select that open drive actor and check it's details tab

19. Click on generate routes button to generate route planner

20. Click on add spwan points

21. Search for CarlaMapGenartor Actor and dropped it on the map.

22. Search for vehicle spawn spoint and put it on one of the box generated by route planner

23. Search for player start and put it on one of the box generated by route planner

24. Search for : AtmosphericFog, BP_Sky_Sphere, DirectionalLight, skyLight and add it on the map

25. Click on play button

26. Run Manual_control.py
>

closed 2019-05-24 by Dhairyashil-kpit

NebulousNeo, 2019-09-05:
I found a solution, it was the fault of the .egg file in the folder there wasn't the right .egg file version (carla-0.9.5-py2.7-linux-x86_64.egg instead of should be carla-0.9.6-py2.7-linux-x86_64.egg). This problem occur because i had two carla folder the first one was the old carla version(0.9.5) and the new one(0.9.6 build from source). So i copied from folder /home/xxx/Test/carla/PythonAPI/carla/dist/ (folder of 0.9.6) the egg files to /home/xxx/yyy/zzz/Carla/CARLA_0.9.5/PythonAPI/carla/dist (this is 0.9.5 where i also execute my pythonAPI). after that it worked and i could conntect to the carla world.

aurthes6220, 2019-09-24:
>     16\. It will generate map in view port

I followed steps until 16.It will generate map in view port.

....then crashed.

whatgit, 2019-10-29:
> 19. Click on generate routes button to generate route planner

This does nothing for me, i.e. it did not generate any route planner. I have my own OpenDrive and .fbx files, which are NOT generated by RoadRunner. UE did not crash when I press "play" to play the level though.

zc804351220, 2020-02-28:
can you tell me which sofrware did you used to generate .fbx files？thank you

dannymanastireanu, 2020-08-17:
> @whatgit
> can you tell me which sofrware did you used to generate .fbx files？thank you

Road Runner from VectorZero( now Matlab)

athulg, 2021-01-11:
> Hi,
>
> I Finally Got it !!
>
>
> 1. Download a sample map (.xodr : OpenDRIVE format) from : http://www.opendrive.org/download.html
> 2. Open Road Runner application and import the downloaded map
> 3. Export [...]
I tried the above steps and now I still get the same errors and my viewport of Unreal Engine is blank too.

SiZHANG0303, 2023-06-29:
Is RoadRunner a must for importing my map into Carla?