carla-simulator/carla #1663: Customized map does not work.

z1223343, 2019-05-21:
**Hello!**

I just tried to customize map in Carla 0.9.4. based on the default map.
After deleting some traffic light groups, traffic lights and static meshes in Town06, UE Editor could not work with python client normally. After making package, the simulator cannot even found the new map.

I have easily used my own map in Carla before the version 0.9.0, but now I can't.
If Carla is so sensitive to changes, how can I customize the maps?

**Thank you for any suggestion.**

Sakrust, 2019-05-23:
You changed Town06 and saved it as such or did you add a new map with different name?

z1223343, 2019-06-10:
> You changed Town06 and saved it as such or did you add a new map with different name?

Sorry for the late response, I have tried both ways and they all did not work.

DatKartoffel, 2019-08-04:
Hello, have you found the solution? I met the same problem here.

sheelabhadra, 2019-10-23:
Hi @z1223343 and @DatKartoffel, I took a similar approach in Carla 0.8.2 and created a custom map after duplicating an existing map and then deleting objects from the custom map. I packaged it after making changes. Although my custom map works in the editor, the packaged version can't find the map. Were you able to find a way to load custom maps after packaging?

OS: Ubuntu 16.04
Carla version: 0.8.2 (stable)