carla-simulator/carla #1664: run carla not in graphical mode

kimiya66, 2019-05-21:
Is it possible to run the carla (carla-server) through commands and not graphically?
Because I want to access my program remotely and it is not possible that I always open unrealEngine (or run ./CarlaUE.sh) as it needs high graphic card and RAM..so to see the results of my code in python I should run simulator(to connect to carla-server) but I do not want to always open the map and just run simulator in command line without graphical view...

aadityasp1, 2019-05-22:
Try this, it should work..

DISPLAY=:  ./CarlaUE4.sh

kimiya66, 2019-05-22:
> Try this, it should work..
> 
> DISPLAY=: ./CarlaUE4.sh

Hi,
I tried this but this is not working. I typed DISPLAY=: ./CarlaUE4.sh -carla-server in carla dir during remote access but I got the same errors as before(without DISPLAY....). errors about OpenGL and segmentation fault(core dumped)

user025, 2019-05-23:
@kimiya66 It's not possible that you don't use a graphic card for rendering and get the whole world in CarlaUE4...
Try to update your graphic card driver first. Segmentation faults at start are mostly related to driver issues.

kimiya66, 2019-05-23:
> @kimiya66 It's not possible that you don't use a graphic card for rendering and get the whole world in CarlaUE4...
> Try to update your graphic card driver first. Segmentation faults at start are mostly related to driver issues.

That is my quesion. that I do not have access to graphic drivers remotely and I want to access carla without loading graphical env in UnrealEngine.

closed 2019-07-29 by stale[bot]