carla-simulator/carla #1667: [Error]: make package: Failed to load Parent for BlueprintGeneratedClass

JMorceaux, 2019-05-22:
Hi,

I recently changed to UnrealEngine_4.22 and the latest carla commit (ce3ea1c). It appears that the make package command doesn't include the carla content directory. I have this error for each asset.

```
  LogLinker: Warning: CreateExport: Failed to load Parent for BlueprintGeneratedClass /Game/Carla/Blueprints/Walkers/BP_WalkerSpawner.BP_WalkerSpawner_C
````
I think, it is linked to the Carla UE4 project file, but I'm not sure what to change.

Thanks !

nsubiron, 2019-05-22:
Hi @JMorceaux,

Have you updated the assets with `Update.sh`?

JMorceaux, 2019-05-22:
Hi @nsubiron, yes I did a full re-download (UnrealEngine and Carla) to make sure I have a clean compilation.

nsubiron, 2019-05-22:
Check that the assets downloaded correctly, make sure the folder `Unreal/CarlaUE4/Content/Carla` is not empty, it should be about 7,7GB in size.

closed 2019-05-22 by JMorceaux

JMorceaux, 2019-05-22:
Thanks for the assistance @nsubiron !