carla-simulator/carla #1668: Build error on Windows

thomasrantian, 2019-05-22:
When I build using x64, I got the thefollowing error:
"""
C:\Program Files\carla\carla>make rebuild                                                                               -[Setup]: [Batch params]: --boost-toolset msvc-14.1                                                                     -[Setup]: Asynchronous jobs:  8                                                                                         -[Setup]: Boost toolset:      msvc-14.1                                                                                 -[Setup]: Install directory:  "C:\Program Files\carla\carla\Build\"                                                     -[Setup]: Installing zlib...                                                                                                -[install_zlib]: [Batch params]: --build-dir "C:\Program Files\carla\carla\Build\"                                  Files\carla\carla\Build\""=="" was unexpected at this time.                                                             make: *** [setup] Error 255        

"""
Any ideas what is the issue?

Sakrust, 2019-05-23:
Probably the wrong compiler. Try updating visual studio to 2017 and removing all old versions. I guess you are using version 2015.

thomasrantian, 2019-05-23:
I am using VS2017...

closed 2019-07-29 by stale[bot]

DamnRicky, 2019-10-02:
any solution?

JJW152, 2019-11-02:
I am having this same problem. I desperately need some help here.

The most frustrating part is I installed Carla on this machine before. Uninstalled, how I can't do anything because of this error.

I have reinstalled VS 2017. Nothing.

anurag-176, 2019-11-29:
Put the "carla" Folder that you cloned from git into your C drive
Then execute the command "make launch" from the x64 VS CLI by setting the directory into "C:\Carla"
But make sure you have "make", "cmake" and other dependencies listed [here](https://carla.readthedocs.io/en/latest/how_to_build_on_windows/) installed.

OlgerSiebinga, 2020-04-06:
Putting carla directly in C drive works for me, it seems like having space characters in the path when calling make is the issue.