carla-simulator/carla #1669: Is there any API equivalent to deprecated 'client.read_data()'

kouyoumin, 2019-05-23:
I need to log all vehicles and their status (transform, velocity, control...) at every moment.
This how I implement:
```
while logging:
    timestamp = world.wait_for_tick()
    vehicles = self.world.get_actors().filter('vehicle.*')
    for vehicle in vehicles:
        transform = vehicle.get_transform()
        velocity = vehicle.get_velocity()
        control = vehicle.get_control()
        log.append('%d,%f,%f,%d,%s,%f,%f,%f,%f,%f,%f,%f,%f,%f,%f,%f,%f,%r,%r,%r,%d\n' % (timestamp.frame_count, timestamp.elapsed_seconds, timestamp.delta_seconds, vehicle.id, vehicle.type_id, transform.location.x, transform.location.y, transform.location.z, transform.rotation.pitch, transform.rotation.yaw, transform.rotation.roll, velocity.x, velocity.y, velocity.z, control.throttle, control.steer, control.brake, control.hand_brake, control.reverse, control.manual_gear_shift, control.gear))
```
However, I found that data may come from different ticks but not the timestamp I expect.
Say, I may get timestamp from frame 1, transform from frame 2, velocity from frame 3...
How do I correctly get all data at SAME time?

Sakrust, 2019-05-23:
Use synchronous mode and fixed-step solver. Both can be found in their documentation. The you need to add a world.tick() inside your while loop, which advances the world exactly one frame every step.

kouyoumin, 2019-05-24:
I tried and it works like a charm.
Thank you so much.

closed 2019-05-24 by kouyoumin

HarnarayanSingh, 2020-05-25:
@kouyoumin Can you tell how you managed to do this? I'm looking to log similar data from simulation.

kouyoumin, 2020-05-25:
@HarnarayanSingh Try this:
```
world = client.get_world()
settings = world.get_settings()
settings.synchronous_mode = True
world.apply_settings(settings)

world.tick()
while logging:
    timestamp = world.wait_for_tick()
    vehicles = world.get_actors().filter('vehicle.*')
    for vehicle in vehicles:
        transform = vehicle.get_transform()
        velocity = vehicle.get_velocity()
        control = vehicle.get_control()
        log.append('%d,%f,%f,%d,%s,%f,%f,%f,%f,%f,%f,%f,%f,%f,%f,%f,%f,%r,%r,%r,%d\n' % (timestamp.frame_count, timestamp.elapsed_seconds, timestamp.delta_seconds, vehicle.id, vehicle.type_id, transform.location.x, transform.location.y, transform.location.z, transform.rotation.pitch, transform.rotation.yaw, transform.rotation.roll, velocity.x, velocity.y, velocity.z, control.throttle, control.steer, control.brake, control.hand_brake, control.reverse, control.manual_gear_shift, control.gear))
    world.tick()
```
Be aware the code above is old and may not compatible with latest API.

HarnarayanSingh, 2020-05-25:
@kouyoumin  Thanks for the reply. Could you tell which Carla version did you try this out with?

kouyoumin, 2020-05-26:
@HarnarayanSingh Possibly 0.9.5