carla-simulator/carla #1671: Adding new APIs, API Attributes definition.

vishal-prgmr, 2019-05-23:
Greetings,

Where are the attributes of each **sensor measurement** for the PythonAPI defined? For example, LidarMeasurement has the attributes **.raw_data, .channels, .horizontal_angle.** 

Where are these defined? I want this information as I am looking forward to add an API which will add in semantic segmentation of the PointCloud.

user025, 2019-05-24:
@vishal-prgmr Original structure is described in libcarla/source/carla/sensor/s11n/LidarSerializer.h and the definition in PythonAPI/carla/source/libcarla/SensorData.cpp

You can check it by grep keyswords like LidarMeasurement in those Source folders.

[the build system document of carla](https://carla.readthedocs.io/en/latest/build_system) describes how the system works. It may also give you some clues.

Atlunch, 2019-05-28:
Hello everyone,

i would like to set the camera saturation value to get a gray scale image.

@user025 thank you for the information....but i didn't get it.

Is Carla using the standard unreal camera actor?
How can I add a new attribute? Like for example "Aspect Ratio" or "Saturation"

I do not understand the connection between all these different libcarla and c++ scripts.

Could someone explain a way of adding a new camera attribute by the example of implementing the "saturation"?

analog-cbarber, 2019-05-28:
I don't believe the API allows you to specify parameters of the cameras other than those described in the documentation. The aspect ratio is just the ratio of the image width and height, which you can set. If you want grayscale video, the easiest thing to do would be to just compute it from the RGB values. Otherwise, you might have to add your own camera sensor.

closed 2019-05-29 by vishal-prgmr

Atlunch, 2019-05-29:
@analog-cbarber "Otherwise, you might have to add your own camera sensor."
That would be interesting too!

How could i add my own camera sensor?

analog-cbarber, 2019-05-29:
You really should just look at the existing camera sensors to see how they are done. The ones that I have done involved creating new post processing materials in the Unreal Editor.

I really wouldn't bother unless what you want cannot be done with the existing sensors or you really need the extra speed.