carla-simulator/carla #1673: How to run a car on customized route and perform lane changing at a given point?

Derekabc, 2019-05-23:
HI, is there away for me to design a route(setup a waypoint?) and let the car follow predefined route and does lane changing at a given time?   Thanks for any suggestions!

krdindorkar19, 2019-06-26:
@Derekabc Did you find any way to implement it ?

Derekabc, 2019-06-26:
you can study the code automatic_control.py. it use the globalRoutePlanner to design a route(return a list of waypoints), and it controls the vehicle by PID. I do Lane Changing by indicating the new waypoints on the next lane when there is a car ahead of the ego car within a threshold. Maybe you need to find a good parameter for the PID.

krdindorkar19, 2019-06-27:
@Derekabc  Hi, thank you for the information, can you help me with the in detailed description of how to generate route using global planner

Derekabc, 2019-06-27:
hi, you can generate a list of waypoints from point A to Point B by code like this:
```
from agents.navigation.global_route_planner import GlobalRoutePlanner
from agents.navigation.global_route_planner_dao import GlobalRoutePlannerDAO
dao = GlobalRoutePlannerDAO(map, 2.0)
        grp = GlobalRoutePlanner(dao)
        grp.setup()

a = carla.Location(x=96.0,
                           y=4.45,
                           z=0)
b = carla.Location(x=215.0,
                           y=6.23,
                           z=0)
w1 = grp.trace_route(a, b)
```
You can find more details in automatic_control.py.

closed 2019-09-02 by stale[bot]

SamiaAbid, 2019-12-06:
@Derekabc where did you write this code? can you please guide me?

Derekabc, 2019-12-06:
@SamiaAbid , [code](https://github.com/Derekabc/RL/blob/master/Carla_HM/lane_changing-v2.py).  You can just take it as an example, as our settings are different.
In the game_loop() function, I did one thing, if there is a vehicle ahead of us and distance is within a threshold, then the ego car does a lane changing.
Hope it is helpful!

jizefeng0810, 2020-10-17:
> @SamiaAbid , [code](https://github.com/Derekabc/RL/blob/master/Carla_HM/lane_changing-v2.py). You can just take it as an example, as our settings are different.
> In the game_loop() function, I did [...]

Hi @SamiaAbid, I want to generate a route map and let the vehicle drive along the route. What should I do? Do you have a script to provide?

Zhuweilong123, 2021-02-19:
How can I use PID to control the ego-vehicle according to a route (a list of waypoints returned by  globalRoutePlanner )? I really hope you can give me some help. Best wish! @Derekabc

omaarelsherif, 2021-07-23:
Hi @Derekabc 
Can you explain how can I perform lane changing?