carla-simulator/carla #1680: Problem with generate_map.py

eloy96vg, 2019-05-26:
Hi,
I have tried to generate the map with generate_map.py and the map is much smaller than routeplanners, spawn points, etc(it is observed in the pictures). I'm using a map created in RoadRunner.
Someone knows how to solve this problem?

[image]
[image]

Thanks.

dnovillo-CVC, 2019-05-28:
Hi!
It sometimes happens, it is UE4 related afaik. There is two possible routes to fix this:

1. Erase the map and regenerate it with generate_map.py. We don't know yet which are the exact conditions why this happens, but it sometimes is hard to overcome this on a specific computer, we don't yet know why as the problem happens from time to time and it is hard to reproduce sometimes.
2. If that's not working, selecting the generated actors and setting the scale to 100 should do the trick. You might need to rotate the pieces also.

Let me know if that solves the issue.

Cheers!

eloy96vg, 2019-05-29:
Hi,

I have solved the problem setting the scale to 100.

Thanks for your help @dnovillo-CVC !

closed 2019-05-29 by eloy96vg