carla-simulator/carla #1681: Carly 0.9.5: is there afixed target velocity?

slinshady89, 2019-05-27:
referring to  #1113 there was nothing like a fixed velocity setting in 0.9.1. Is this still true? 
Is it somehow possible to set a fixed velocity or is it possible to use street signs as advisors so I just change every existing one to a 30 traffic sign instead it's original state?

michael-good, 2019-05-28:
It is possible: vehicle.set_velocity(8.3) #sets speed to 30 km/h (8.3 m/s)
However, this solutions seems to break autopilot. It is useful only if you have implemented a control system.
On the other hand, you can delete 90 km/h and 60 km/h speed limit signs. If you do this, the vehicle has a fixed velocity of 20 km/h with autopilot enabled. Here is how:
```
for speed_sign in world.get_actors().filter('traffic.speed_limit.*'):
    if speed_sign.type_id == 'traffic.speed_limit.90' or speed_sign.type_id == 'traffic.speed_limit.60':
        speed_sign.destroy()
```

slinshady89, 2019-05-29:
yeah I already tried and saw the behavior that he's driving 20kp/h. 

But the other thing is maybe useful when I implemented my path planner that publishes waypoints for the ros_bridge_ackermann_controller. Thanks a lot

closed 2019-05-29 by slinshady89

nobodywu, 2019-12-04:
Carla 0.9.6, How to set velocity in 3D vector?
[carla actor class](https://carla.readthedocs.io/en/latest/python_api/#carlaactor-class), set_velocity method need a 3D vector