carla-simulator/carla #1683: LIDAR lane markings annotated data

rfsch, 2019-05-27:
Hello 

I would like to use Carla with LIDAR sensor information. Is there annotated road marking data available in pointcloud format. I.e. is it possible to have together with a Lidar frame (xyz data) ground thruth of the road boarder, lanes together in xyz format ? 

Thanks

user025, 2019-05-28:
Currently the official version doesn't support this.

There are two ways to realize it.

1. Use synchronous mode and check if a point is in a bounding box of vehicles(using the location of previous frame). Otherwise it is a ground point. There are some known issues related to this,  like #1647, #1312  . So you may have to do some adaption.

2.  If you are familiar with the source code, you can get the actor id when lidar sensor hit a object. The actor id is available via

`auto actor_temp = HitInfo->GetActor(); `
`auto actor_id = GetEpisode().FindActor(actor_temp).GetActorId(); `

In carla-0.9.5/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/RayCastLidar.cpp

Or you may check my forked branch at mul_t

rfsch, 2019-05-28:
Thank you for your answer. But how would it be possible to get in this way Lane and road border markings. I do not want to have to detect or discover them programmatically. I would need to have the information about them in xyz in order to reason about them.

Thanks

vishal-prgmr, 2019-05-28:
@user025 , i can now obtain a pointwise labelled pointcloud from the source code. However trasmitting the labels through the boost buffer has been creating a havoc. I see that the pointcloud is offset buffered and accessed using the frombuffers method of numpy. 

Have you also tried buffering some new information out of the source code?

user025, 2019-05-28:
@rfsch Sorry, seems I don't quite get your question. 

If you are seeking for a groundtruth road, check this [document](https://carla.readthedocs.io/en/latest/python_api/#carlamap).

`save_to_disk`  provides the whole map in opendrive format.

`get_waypoint` provides informations about roads. Example of using get_waypoint API could be found in the no_rendering_mode.py. 

Hope it helps.

rfsch, 2019-05-28:
@user025 thank you for your help!

Sorry if I didn't explain it well.

I would need LIDAR recordings of a car (Carla?) driving along a road. For detection of cars that are oncoming to the car I am driving I am using Lane borders. For these I would need to correct xyz coordinates and knowledge that they are lane borders. 

Could I record the lidar data captured and know which points are borsers and lanes ?

Thanks!

user025, 2019-05-29:
@rfsch Lidar data provides points only. You can get groundtruth of borders and lanes of lidar data either by bounding box checking, as I mentioned in my previous answers, or by manually labeling.

rfsch, 2019-05-29:
@user025 So does it mean that by getting the actor id of an object hit by the lidar sensor I am able know it is a lane, border, etc ? I would need this information while driving and would not want to manually label it. Or is manually labeling easier ?

user025, 2019-05-30:
@rfsch  actor id will always return 0 if lidar hit at anything not an carla actor, etc. buildings, grass, road. I believe manually labeling would be easier.

rfsch, 2019-05-30:
@user025 thanks. That was what I was afraid of. How would I manually label in Carla ?

barbierParis, 2019-06-20:
https://github.com/ZhuangYanDLUT/carla have you checked this ?

rfsch, 2019-06-20:
@barbierParis Thanks! This looks good.

barbierParis, 2019-06-21:
@rfsch if you use this forked version, would you mind giving some feedback on the results ? I'm currently having issues running CARLA on linux so I haven't been able to try it.

closed 2019-08-27 by stale[bot]