carla-simulator/carla #1686: Displays gibirish when asked to print(time)

Abanoub-G, 2019-05-28:
Hi,
I have a question that might seem a bit basic but I can't find an answer for it.
When I write: 
current_timestamp = carla.Timestamp.delta_seconds
print(current_timestamp)

in python I get this:  
<property object at 0x7f587800a470>
displaying in my terminal. I understand that this is some kind of encryption to for the server to use to communicate with unreal. But what if I actually want to display this for debugging purposes. How do i do that? I couldn't find any help in the documentation. 

Thanks

analog-cbarber, 2019-05-28:
Your code is printing out the property attribute on the Timestamp class. It has nothing to do with communicating with the server. 

You want a Timestamp instance. You can use world.wait_for_tick() to get a timestamp. There are also timestamps in returned sensor data.

Abanoub-G, 2019-05-28:
Hi @analog-cbarber ,

I didn't quite get what you mean? How would that be written in code?

analog-cbarber, 2019-05-28:
Assuming you have already attached a client, I think it would be something like:

~~~python
>>> world = client.get_world()
...
>>> t = world.wait_for_tick()
~~~

Abanoub-G, 2019-05-28:
Hi @analog-cbarber 
Yes, thanks a lot that did work. I thought world.wait_for_tick() pauses the simulator (UE4) until the python script runs it again but apparently I miss understood it. 

In this case how would I get the location of a certain agent in the spawn_npc.py example? Would it also be linked to world.wait_for_tick()?

What I have tried is : 
```
while True:
            actor = actor_list[0]
            actor_location = actor.get_location()
            print(actor_location)
```

but I get an error 
    actor_location = actor.get_location()
    AttributeError: 'int' object has no attribute 'get_location'

Any idea what I am doing wrong?

analog-cbarber, 2019-05-28:
I believe that wait_for_tick only pauses the simulator in synchronous mode. 

Your actor list contains integer actor ids instead of Actor objects.  I think that is fairly obvious from the spawn code. You can use the id to look up the corresponding Actor object from  the world object.

I get the impression that you are not entirely familiar with python. You probably should spend some time learning the language so that you can better understand the examples.

analog-cbarber, 2019-05-28:
In any case, it is pretty clear this is not a bug.

Abanoub-G, 2019-05-29:
Thanks, I never said it was a bug. Well I am certainly not familiar with carla's context, but that doesn't necessarily mean that I am not familiar with python. 

Anyway, according to what you said the below code should make it work. However, I am still getting an error. Any idea what I am doing wrong?
```
agent_location  = world.get_actors(actor_ids=actor_list[0])
```

Thanks

analog-cbarber, 2019-05-29:
Sorry, I thought you were not that familiar with python because of the nature of your questions. E.g., you were trying to print an attribute of a class object, which is not something I would expect an experienced python programmer to do.

In any case, you didn't say what error you are getting, but I am not aware that get_actors takes any arguments. You really should spend some more time reading the Python API tutorial and looking at the example code.

Abanoub-G, 2019-05-29:
Yes, I have been looking through those for a while now and I have managed to make it get all of the actors at once, but what I was trying to do is to only get the agent I am looking for using a certain id. 

I have now managed to solve my problem and get the specified agent details rather than getting the details of all of the actors. I'll explain the problem, just to clarify that it wasn't a problem with me not doing enough background reading. 

The problem was that I was getting the world before my while true loop and trying to get the actors ids which did not even exist and this I still don't understand; because as soon as I spawn them before the while true loop I would expect them to exist in the world. This might be a cause of being in the synchronous mode. However this doesn't come into action until the second run in the while true loop is executed. My usage of the API was correct all along, it was just that the world i was getting did not have the agents in them yet, so the ids I was supplying didn't make any sense.  

Anyway thanks for your help.

closed 2019-05-29 by Abanoub-G