carla-simulator/carla #1687: GetPointCount and get_point_cloud

vishal-prgmr, 2019-05-28:
Greetings,
 
**.channels** returns the number of channels. 

What does **get_point_count** return in LidarMeasurement?

ThankYou

nsubiron, 2019-05-28:
Hi @vishal-prgmr,

`get_point_count` returns the number of points for the given channel
```py
for x in range(0, lidar_measurement.channels):
    acc += lidar_measurement.get_point_count(x)
assert acc == len(lidar_measurement)
```

vishal-prgmr, 2019-05-28:
@nsubiron , just figured this and now its verified. Thank You for your time. :)

vishal-prgmr, 2019-05-28:
@nsubiron , my buffer has something else coming right after apart from the pointcloud and the header. 

I assume if I offset the buffer with an amount equal to the **acc** (number of points of the pointcloud), I would be able to access the extra information , right?

closed 2019-05-29 by vishal-prgmr