carla-simulator/carla #1688: Pausing simulator

Abanoub-G, 2019-05-28:
I need the simulator to pause while some other piece of script runs which communicates with another piece of software. 

From the discussion in issue #1274 I understood that UE4 will wait for the python script to execute world.wait_for_tick() or world.tick() in order for it to run the next tick in the simulator (UE4). 

I have done a little test, which is:
```
while True:
            world.wait_for_tick()

            time.sleep(10)
```
I was expecting that the simulator will wait 10seconds between each tick, but this didn't happen. Any idea why it is not doing what I am expecting?

Thanks

nsubiron, 2019-05-28:
You need to enable synchronous mode and use `world.tick()` to step the simulation, take a look at the example [synchronous_mode.py](https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/synchronous_mode.py).

Abanoub-G, 2019-05-29:
Thanks a lot @nsubiron . That solved it.

closed 2019-05-29 by Abanoub-G

Nitro60zeus, 2023-08-26:
> Thanks a lot @nsubiron . That solved it.

Can you please share your code to pause simulation