carla-simulator/carla #1695: Carla 9.5 : Import an Articulated Vehicle in CARLA

Dhairyashil-kpit, 2019-05-30:
Hi,

I am new to CARLA, Unreal Engine and 3D Modelling Software (blender). 
**I want to Import an Articulated Vehicle (Truck + Trailer) into CARLA.** 

**For newbie, the Carla's documentation is very difficult to understand so I want to validate following process of importing Articulated Vehicle in CARLA.**

**Process to import Articulated Vehicle in CARLA:**
    **1. 3D Modelling Software (Blender):** blend file with desired bone structure. 
    **2. Unreal Engine:** Generate skeletal mesh, generate blueprints, etc.
    **3. CARLA:** place vehicle files into Carla's directory structure. 

[image]

**Steps 1:**  3D Modelling Software (Blender) : 
  
              1]  A Vehicle Model’s mesh
                    i. Bone structure in the model as follow:              
                       a. There should be a root bone attached to the center of the vehicle.
                       b. And other bones are attached to each of the components(Wheels, Joint of trailer) 
                       c. Bones of all the components are attached to the root bone
                       d. Refer Issue : https://answers.unrealengine.com/questions/402237/vehicle-skeletal-mesh-import-problem.html
                       e. Refer :  https://www.youtube.com/watch?v=9gdmJIfuORk
                    ii. Export the file into .fbx format

**Step 2:** Unreal Engine :  

               1]   
                    a. Import .fbx file as a Skelletal Mesh
                    b. Delete Automatically generated physics assets
                    c. Add custom Physics Assets :
                        1. Create the Body Collision
                        2. Create the Wheel Collision
                    d. Refer Vehicle Art Setup in Unreal Engine 4 : https://docs.unrealengine.com/en-US/Engine/Physics/Vehicles/VehicleContentCreation/index.html

              2] 
                  a. Creating wheel Blueprints :
                       1. It will need Wheel's Radius and Width

              3] 
                  a. Creating a Vehicle Animation Blueprint

              4] 
                a. Creating a Vehicle Blueprint :
                     1. TireConfig Data Assets, Wheel Blueprints, and an Animation Blueprint
                     2. Needs to bring together to create a fully functional vehicle in Unreal Engine 4 (UE4). 
                     3. Add Camera Component
                     4. Add Vehicle Movement Component : Refer https://carla.readthedocs.io/en/latest/how_to_model_vehicles/ 
                          - Set Wheel Class
                          - Set Bone Name

              5]  Setup CARLA Game Mode

**Steps 3:** Carla: 

               a. I refer Carla related configuration from https://carla.readthedocs.io/en/latest/how_to_add_assets/

**References:** 

 1. Refer Vehicle User Guide in Unreal Engine 4 : https://docs.unrealengine.com/en-US/Engine/Physics/Vehicles/VehicleUserGuide/index.html

2. Refer similar process as described in Vehicle User Guide with respect to CARLA : https://carla.readthedocs.io/en/latest/how_to_add_assets/#adding-a-vehicle

3. Blender to Unreal : https://www.youtube.com/watch?v=9gdmJIfuORk

4. https://answers.unrealengine.com/questions/402237/vehicle-skeletal-mesh-import-problem.html

closed 2019-08-07 by stale[bot]

Dhairyashil-kpit, 2019-10-24:
Hi all,

Any one tried to import sample vehicle into CARLA_9.6 ?

If any one have documented video / steps of importing vehicle design for generation fbx with bone structure and import process of fbx into carla, then please share with us.

Thanks in Advance.
Dhairyashil

RonGros, 2020-03-04:
+1 - Were you able to add the truck to Carla? We also want to try that....

RonGros, 2020-03-05:
Hi Dhairyashil,
Thanks! that would be very helpful! 
Appreciate it!
Ron

RonGros, 2020-03-10:
Hi @Dhairyashil-kpit  - it will be great if you can share the information

onebitme, 2020-04-24:
@Dhairyashil-kpit @RonGros Dear fellows,
Do you have any updates on this topic? Were you able to put a truck and trailer model in CARLA?

Thanks!

Dhairyashil-kpit, 2020-04-24:
Hi,

https://continuebreak.com/articles/how-rig-vehicle-blender-28-ue4/

Follow this article to create dummy vehicle in blender and model into CARLA, tried in CARLA9.6
also you need to understand all the steps while importing FBX file into CARLA, so it will help you to debug issues which will occur in import process.

Note:

  1.  The above article will help you to model dummy vehicle, If you want to create on vehicle model in blender and model into CARLA you have to understand the "HOW TO SET BONE STRUCTURE TO VEHICLE MESH DESIGN" also the physics model of 4-wheeled vehicle of CARLA.
  2.  While Tuning the bounding box for vehicle model CARLA UE4 will crash with "Core Dump/Segmentation fault"
  3.   I tried to find on ready-made FBX file with vehicle mesh design and proper bone structure attached to it, but not success-ed init, so it will become easy to focus the only on vehicle FBX import process in CARLA.

Regards,
Dhairyashil

This message contains information that may be privileged or confidential and is the property of the KPIT Technologies Ltd. It is intended only for the person to whom it is addressed. If you are not the intended recipient, you are not authorized to read, print, retain copy, disseminate, distribute, or use this message or any part thereof. If you receive this message in error, please notify the sender immediately and delete all copies of this message. KPIT Technologies Ltd. does not accept any liability for virus infected mails.

onebitme, 2020-04-24:
> Hi, https://continuebreak.com/articles/how-rig-vehicle-blender-28-ue4/ Follow this article to create dummy vehicle in blender and model into CARLA, tried in CARLA9.6 also you need to understand all [...]

Dear @Dhairyashil-kpit I've read the article you had shared and it is definitely a good point to start modeling the vehicle itself.

However I was not able to find or maybe understand how it enables us to put an articulated vehicle inside the CARLA

Do you also have any resources for that? Kind regards!

Dhairyashil-kpit, 2020-04-24:
Hi,

sorry for not able to share the total with you guys for importing  articulated vehicle. [pardon my english.] 😄

first try to understand:

  1.  create FBX file vehicle mesh and Bone structure attached to it for 4 wheel vehicles.
  2.  Then import process in UE4 then CARLAU4 for dummy vehicle with available 4 wheel vehicle blueprints. and try to use that vehicle in CARLA scenarios.
  3.  Then create FBX file for  articulated vehicle with proper bone structure attached to it [need help of professional blender designer to create such vehicle design and set proper Bone structure to trailer, main truck, and 5th wheel of trailer].
  4.  Then to import articulated vehicle. we need try to attach the existing vehicle blueprint and if possible create own blueprint for articulated vehicle.

 Note:

  1.  For articulated vehicle, there are some experimental steps like:
     *   articulated vehicle with only 4 wheels are enabled or bone structure attached to 4 wheels only, so it will look like articulated vehicle and only 4 wheel are enabled.
     *   We can set proper bone structure to articulated vehicle, and generate the FBX file, but before importing it into CARLA directly try it to import in UE4.
     *   then what to support the movement of articulated vehicle in carla we need to update the available blue prints.

     2. To achieve importing for articulated vehicle in CARLA, we need to understand How to import it
         into UE4, then try to play that vehicle, then jump into CARLA.

    3. This is my understanding, but it will need to much time import articulated vehicle in CARLA but
        after importing dummy vehicle you will understand what things are available in carla to
        import articulated vehicle.

kindly share your experience with US.

Thanks and Regards,
Dhairyashil

________________________________ From: Ersun Sözen notifications@github.com<mailto:notifications@github.com> Sent: 24 April 2020

RonGros, 2020-04-28:
> @Dhairyashil-kpit @RonGros Dear fellows,
> Do you have any updates on this topic? Were you able to put a truck and trailer model in CARLA?
> 
> Thanks!

Hi @onebitme - sorry for not replying so far.
TBH I delegated this to another member (he is working off-shore) and he was able to do it according to the carla documentation 
I think he used this:
https://carla.readthedocs.io/en/latest/tuto_A_add_vehicle/

and this:
https://docs.unrealengine.com/en-US/Engine/Physics/Vehicles/VehicleUserGuide/index.html

I don't have more information at the moment, hope it helps a bit