carla-simulator/carla #1700: Carla cannot load Town01 (Ubuntu - carla_0.9.5)

willruiz, 2019-05-31:
I have been using the documentation to load the maps that come with carla with the command 
$ ./CarlaUE4.sh /Game/Maps/Town01
and I get the following message 
"The map specified on the commandline '/Game/Maps/Town01' could not be found. Would you like to load the default map instead?"
> image of error: https://imgur.com/a/kZh6MPY

I just downloaded the carla 0.9.5 release and I have not moved any of the maps around
I tried to access the map using another directory path which seems to be more accurate
$ ./CarlaUE4.sh /CarlaUE4/Content/Carla/Maps/Town01.umap
but this just resulted in the a segmentation fault.

Can someone explain how to run Town01 or any map from the 0.9.5 build? I'm really lost. 
Thanks!

user025, 2019-05-31:
Try `./CarlaUE4.sh /Game/Carla/Maps/Town01`

willruiz, 2019-05-31:
Yep, that worked. Thanks for the quick fix!

closed 2019-05-31 by willruiz

CMakey, 2023-12-27:
> Try `./CarlaUE4.sh /Game/Carla/Maps/Town01`

hi bro, I'm also meeting this question too.

but Carla couldn't find maps directory in 0.8.4 in ubuntu , do u had met this problem before?

waiting for u reply and more than appreciated for it.