carla-simulator/carla #1706: ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused --->not a single .py file is operating

shujatalikhan, 2019-06-01:
HELLO , I am trying to find answer but i am not getting it ,is there any solution ??
please help!

shujat@shujat-Inspiron-7570:~/carla/CarlaSimulator/PythonClient$ python client_example.py 
INFO: listening to server localhost:2000
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused

closed 2019-06-07 by shujatalikhan

omar-Mohamed300, 2019-11-28:
can you please tell me how you managed to solve this problem?

meteorshowers, 2020-02-05:
the same!
why close???

Jidenna, 2020-04-27:
Having the same problem here