carla-simulator/carla #1707: getting RuntimeError: rpc::rpc_error during call in function version when trying python examples

CodeTrol, 2019-06-02:
I'm running a docker version of carla on my Ubuntu server with a gpu in my local network. It consumes around 2719 MB VRAM and doing some calculations (according to nvidia-smi and htop), so I guess it's running the simulation. The docker container runs by this command:
```
docker run -p 2000-2002:2000-2002 --runtime=nvidia -e NVIDIA_VISIBLE_DEVICES=0 carlasim/carla:0.9.5
```

I've cloned the git repo with exact version of carla (0.9.5) locally for creating python clients.
I've installed all dependencies (.deb packages) by following the install docs.
In the carla's root directory, I did:
```
make setup
make PythonAPI
```
got a success code (0x0) in running both commands. According to your instructions how to import carla in python, I need to specify the egg file matching the python version (for a first try, I used 2.7) in order to have no issues importing carla. But I would like to see the existing examples in action and they contain the lines doing that. I'm referring to those lines:
```python
try:
    sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass
```

so, in `<carla root>/PythonAPI/examples`, when I'm trying an example, for example, manual_control.py, I'm getting this error:
```
Traceback (most recent call last):
  File "manual_control.py", line 861, in <module>
    main()
  File "manual_control.py", line 853, in main
    game_loop(args)
  File "manual_control.py", line 774, in game_loop
    world = World(client.get_world(), hud, args.filter, args.rolename)
RuntimeError: rpc::rpc_error during call in function version
```

In all examples, I tried, I got same error at `client.get_world()`.

The OS of the machine where I'm trying to run clients is Ubuntu 18.04.
The OS of the host server is Ubuntu 16.04 but I don't think the issue is with the server (or could be in the carla docker?).
Carla version: 0.9.5

```python
client = carla.Client(args.host, args.port)
client.set_timeout(2.0)
```
^ The above lines don't fail, so not a network issue?

Any ideas? I'm a newbie to this project but it looks promising to contribute to for me.

twojtasz, 2019-06-04:
@CodeTrol I had the same issue running everything locally.  I switched to Python 2.7 and it worked.

AravindChandradoss, 2019-06-05:
I also had this issue when I ran _compiled_ version of Carla. 
By any chance, did you guys add `export PYTHONPATH=$PYTHONPATH:/home/autoware/Aravind/carla/PythonAPI/carla/dist/carla-0.9.5-py2.7-linux-x86_64.egg` to .bashrc?  if so remove it and try. 

Let me know if that works. For running compiled version, It worked for me.

CodeTrol, 2019-06-08:
Sorry for the delay, guys.

@twojtasz I've mentioned already that I'm using Python 2.7. Not sure what else I should switch.
Running `python --version` returns:
```
Python 2.7.15rc1
```

@AravindChandradoss I didn't add anything to `.bashrc`. Here is a tailing of that file:
```
# enable programmable completion features (you don't need to enable
# this, if it's already enabled in /etc/bash.bashrc and /etc/profile
# sources /etc/bash.bashrc).
if ! shopt -oq posix; then
  if [ -f /usr/share/bash-completion/bash_completion ]; then
    . /usr/share/bash-completion/bash_completion
  elif [ -f /etc/bash_completion ]; then
    . /etc/bash_completion
  fi
fi
```
^ everything that comes with Ubuntu.

All I did was: (in the root directory of a cloned carla.git)
```sh
make setup
make PythonAPI
cd PythonAPI/examples
python manual_control.py --host=192.168.1.220 # <-- the host in my local network running a docker version of carla
```

I might look into C++ sources as the error isn't even clear for me.

CodeTrol, 2019-06-08:
I've found that the error is coming from the running carla docker container. I've traced this via wireshark:
[image]
So, running carla in the way I did using a docker version of carla isn't possible?

Both docker container and client are using 0.9.5 of carla.

CodeTrol, 2019-06-09:
I pulled PythonAPI from the docker container and copied to my local machine. No issues using carla python module. Examples are working fine too:
[image]
I guess that's it for now.

closed 2019-06-09 by CodeTrol

asharalam11, 2019-07-17:
Hi, I am still getting the same error even though I have built from source......any help????

adi-geek2, 2019-08-02:
There was a way to stop getting RuntimeError: rpc::rpc_error by ensuring 

a.Same version of Python on all machines in n/w
b.Same version of CARLA on all machines in n/w
It worked for me, can someone please confirm that would be great ?

My setup
- Windows local (CARLA 0.9.5- server render machine) - Python 3.7.0
- Linux VM remote (CARLA 0.9.5- PythonAPI client manual_control.py) - Python 3.7.3

CodeTrol, 2019-09-30:
Same build of CARLA + PythonAPI on all machines in my case. Though PythonAPI can be distributed.

MalathiKaliyappan, 2020-08-25:
Hi @CodeTrol  
I am using python 3.6 on Ubuntu 18.04 with CARLA 0.9.5.
While running spawn_npc.py I got this error `RuntimeError: rpc::rpc_error during call in function version`
```
 python3 spawn_npc.py -n 80

destroying 0 actors

done.
Traceback (most recent call last):
  File "spawn_npc.py", line 124, in <module>
    main()
  File "spawn_npc.py", line 70, in main
    world = client.get_world()
RuntimeError: rpc::rpc_error during call in function version
```
Can you please help me to solve this?

CodeTrol, 2020-09-12:
@MalathiKaliyappan You didn't provide enough information how you got the builds, what steps you followed, etc. From my experience, it's enough to follow the build instructions to get everything compiled locally. It works 100%. I stopped using docker builds of carla due to limited functionality that was in the box and started using through the UE4 editor to modify scenes and create some scenarios for agents. Once you hit "play" in the editor, it starts a carla server and you can connect to it through PythonAPI (use examples).
As for the error itself, this is what I tried and faced:

- Docker carla image + locally compiled PythonAPI = that error
- Docker carla image + extracted the compiled version of PythonAPI from the same docker carla image = works, no errors
- Local carla build + local PythonAPI build (on the same machine) = works, no errors
- Local carla build + extracted the compiled version of PythonAPI from the docker carla image = that error, obviously

Any python version difference across the builds will produce errors (same version of python needs to be used). So, if you're trying the build built on Ubuntu 16.04 which has python v3.5, it won't work on Ubuntu 18.04 (due to python v3.6.9 there), so you have to recompile it for your box.

Hope that helps.

tranquangminh98, 2021-05-20:
python 3.7 solved the problem for me