carla-simulator/carla #1714: Question/feature Carla 0.9.X (5) building .pyd for python 2.7 on Windows

Jabkowsky, 2019-06-04:
VERSION : Carla 0.9.5 , Windows experimental version 

My issue is related to the fact that I would like to use IronPython in C# application to control Carla Client Python Api.
To do this I need to build .pyd library to be able to run Client with python 2.7  ( IronPython works only with 2.7 python version)
I have found in your BuildPythonApi.bat comment:
 "rem We don't provide support for py2 **right now**" 
Question:
I hope it mean that it is possible to build carlib .pyd file with python 2.7 on Windows??
Do you have any instruction or tips for me what should I do to properly build .pyd file? 
( build boost python for 2.7 ?? Something else? :} )
I know that there are already build version for Linux ( python 2.7) but haven't seen any for windows.
Thank you for help !

closed 2019-08-10 by stale[bot]

gmnvh, 2019-09-06:
Hey Jabkowsky, did you get your Windows Carla Client working on Python 2.7? I need to do the same thing here because I am running ROS for Windows and it only runs on Python 2.7. My Carla does not have an egg file for it. Not sure how to generate one.

Jabkowsky, 2019-09-09:
No i did not managed to do this,
Basically issue I found is that there is no c++14 compiler to build correct .pyd file from source code ( python 2.7) 
https://wiki.python.org/moin/WindowsCompilers
Visual C++ | CPython
-- | --
14.X | 3.5, 3.6, 3.7, 3.8
10.0 | 3.3, 3.4
9.0 | 2.6, 2.7, 3.0, 3.1, 3.2
If you want to created .egg file you need to compile source c++ library with Visual studio compiler. ( on windows of course ) 
Probably that is why Carla Developers are giving for windows version only .egg file for py36,
i did manage only to build for 3.7 python version.