carla-simulator/carla #1716: automatic_control.py Script execution failed

ax7476, 2019-06-04:
**[Platform/OS]**
pygame in /usr/lib/python2.7/dist-packages (1.9.1release)

numpy in /usr/lib/python2.7/dist-packages (1.11.0)

Linux  4.15.0-50-generic #54~16.04.1-Ubuntu SMP Wed May 8 15:55:19 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

CARLA_0.9.5

**[Problem description]**
The execution automatic_control.py  script prompts the following errors:

^CTraceback (most recent call last):
  File "automatic_control.py", line 675, in <lambda>
    self.sensor.listen(lambda image: CameraManager._parse_image(weak_self, image))
  File "automatic_control.py", line 715, in _parse_image
    self.surface = pygame.surfarray.make_surface(array.swapaxes(0, 1))
  File "/usr/lib/python2.7/dist-packages/pygame/surfarray.py", line 243, in make_surface
    return numpysf.make_surface (array)
  File "/usr/lib/python2.7/dist-packages/pygame/_numpysurfarray.py", line 368, in make_surface
    blit_array (surface, array)
  File "/usr/lib/python2.7/dist-packages/pygame/_numpysurfarray.py", line 437, in blit_array
    surface.get_buffer ().write (data, 0)
IndexError: bytes to write exceed buffer size

ax7476, 2019-06-05:
Using Python 3 instead of Python 2 solves the problem

closed 2019-06-05 by ax7476

najibghadri, 2020-03-15:
I have the same issue with manual_control.py. But it doesn't work with python3 saying no module named 'carla'