carla-simulator/carla #1723: _Float16 not supported

ironWolf1990, 2019-06-04:
Hello ,

I installed all the dependencies and ran :
`make launch`

I get the following error : 
`/home/ironwolf69/carla/Build/llvm-7.0-source/projects/libcxx/include/type_traits:740:56: error: _Float16 is not supported on this target`
`template <>          struct __libcpp_is_floating_point<_Float16>    : public true_type {};
`

Can someone guide me on this ?? 
Thanks.....

cassfalg, 2019-07-17:
I just encountered the same issue on Arch Linux with clang-8 - I've had to create some symlinks for the Util/BuildTools/Setup.sh script to continue. I would suspect that might be an issue. Which version of clang do you use?

ironWolf1990, 2019-07-24:
@cassfalg  , using clang-8

cassfalg, 2019-07-28:
I have not explicitly checked outside of carla but I suspect that clang 7 can not be compiled with clang 8 due to this issue. I solved that problem for me on arch by installing clang 7 from source and compiling it with gcc 9.1. Then I got the carla build scripts to use that clang 7 to install the additional clang 7 and things worked out more or less...

To make it work on arch I had to edit PythonAPI/carla/setup.py because it checks for the distro and adds hardcoded references to gcc-7 includes/libs, and I had to make symlinks for clang 7 since some other skript checks for /usr/bin/clang-7. I will probably try to make a pull request for the setup.py at least. I'm just not really an expert in cmake, nor do I really have the time to dig really deep, otherwise I would try to make more parts of the carla build system a bit more flexible, check more prerequisites and so on...

closed 2019-10-03 by stale[bot]