carla-simulator/carla #1725: how to fix a camera at a specific location?

geekhut, 2019-06-06:
currently I'm modifying manual_control.py, I would like to fix a camera at a traffic junction (top down view), and capture some image data. How do I do it?

ivanlebedev, 2019-06-06:
In camera setup remove "attach_to" and adjust the transformation

geekhut, 2019-06-06:
> In camera setup remove "attach_to" and adjust the transformation

Hi, thank you for the reply. Sorry I'm really new to Carla, could you provide more details? Like which function to change, or which part of the code I should look at... if you have code example it would be greatly appreciated...

geekhut, 2019-06-06:
> In camera setup remove "attach_to" and adjust the transformation

Hi I managed to control the camera now, but I'm not sure how to set the yaw values to make the camera view top down.

geekhut, 2019-06-06:
> > In camera setup remove "attach_to" and adjust the transformation
> 
> Hi I managed to control the camera now, but I'm not sure how to set the yaw values to make the camera view top down.

all done. thanks.

closed 2019-06-06 by geekhut